OrcaSlicer/OrcaSlicer #2043: After running OrcaSlicer, an error is displayed.

mirspan, 2023-09-09:
### OrcaSlicer Version

1.6.6

### OS version

Manjaro Linux 23.0 KDE

### Printer

Ender 5S1

### How to reproduce

Every time you run OrcaSlicer, you will get an error.: Error running JavaScript: Unsupported result type

### Actual results

[image]

### Expected results

After the error message is acknowledged, the program can be used. The error is displayed every time the program is started.

### Project file & Debug log uploads

2023-09-09 05:56:12.774153[Thread 0x00007fb5df3a1600]:Update_dark_mode_flag: switch the current dark mode status to 0 
2023-09-09 05:56:12.777755[Thread 0x00007fb5df3a1600]:begin to show the splash screen...
2023-09-09 05:56:12.885293[Thread 0x00007fb5df3a1600]:loading systen presets...
2023-09-09 05:56:12.892613[Thread 0x00007fb5df3a1600]:[BBL Updater]:Checking whether the profile from resource is newer
2023-09-09 05:56:12.895196[Thread 0x00007fb5df3a1600]:Installing 2 bundles from resources ...
2023-09-09 05:56:12.898043[Thread 0x00007fb5df3a1600]:[BBL Updater]:Performing 8 updates
2023-09-09 05:56:12.898060[Thread 0x00007fb5df3a1600]:	Update(/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Creality.json -> /home/mirekspanko/.config/OrcaSlicer/system/Creality.json)
2023-09-09 05:56:12.902197[Thread 0x00007fb5df3a1600]:	Update(/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Creality/process -> /home/mirekspanko/.config/OrcaSlicer/system/Creality/process)
2023-09-09 05:56:12.902212[Thread 0x00007fb5df3a1600]:copy_directory_recursively "/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Creality/process" -> "/home/mirekspanko/.config/OrcaSlicer/system/Creality/process"
2023-09-09 05:56:12.963473[Thread 0x00007fb5df3a1600]:	Update(/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Creality/filament -> /home/mirekspanko/.config/OrcaSlicer/system/Creality/filament)
2023-09-09 05:56:12.963495[Thread 0x00007fb5df3a1600]:copy_directory_recursively "/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Creality/filament" -> "/home/mirekspanko/.config/OrcaSlicer/system/Creality/filament"
2023-09-09 05:56:12.978882[Thread 0x00007fb5df3a1600]:	Update(/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Creality/machine -> /home/mirekspanko/.config/OrcaSlicer/system/Creality/machine)
2023-09-09 05:56:12.978902[Thread 0x00007fb5df3a1600]:copy_directory_recursively "/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Creality/machine" -> "/home/mirekspanko/.config/OrcaSlicer/system/Creality/machine"
2023-09-09 05:56:13.027966[Thread 0x00007fb5df3a1600]:	Update(/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Custom.json -> /home/mirekspanko/.config/OrcaSlicer/system/Custom.json)
2023-09-09 05:56:13.030420[Thread 0x00007fb5df3a1600]:	Update(/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Custom/process -> /home/mirekspanko/.config/OrcaSlicer/system/Custom/process)
2023-09-09 05:56:13.030447[Thread 0x00007fb5df3a1600]:copy_directory_recursively "/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Custom/process" -> "/home/mirekspanko/.config/OrcaSlicer/system/Custom/process"
2023-09-09 05:56:13.061002[Thread 0x00007fb5df3a1600]:	Update(/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Custom/filament -> /home/mirekspanko/.config/OrcaSlicer/system/Custom/filament)
2023-09-09 05:56:13.061026[Thread 0x00007fb5df3a1600]:copy_directory_recursively "/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Custom/filament" -> "/home/mirekspanko/.config/OrcaSlicer/system/Custom/filament"
2023-09-09 05:56:13.099259[Thread 0x00007fb5df3a1600]:	Update(/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Custom/machine -> /home/mirekspanko/.config/OrcaSlicer/system/Custom/machine)
2023-09-09 05:56:13.099280[Thread 0x00007fb5df3a1600]:copy_directory_recursively "/tmp/.mount_OrcaSlqIX7bS/resources/profiles/Custom/machine" -> "/home/mirekspanko/.config/OrcaSlicer/system/Custom/machine"
2023-09-09 05:56:13.112417[Thread 0x00007fb5df3a1600]:initialize_network_module, error, dlerror is /home/mirekspanko/.config/OrcaSlicer/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
2023-09-09 05:56:13.112443[Thread 0x00007fb5df3a1600]:initialize_network_module, can not Load Library for /home/mirekspanko/.config/OrcaSlicer/plugins/libbambu_networking.so
2023-09-09 05:56:13.112449[Thread 0x00007fb5df3a1600]:on_init_network: on_init_network, load dll failed
2023-09-09 05:56:13.112454[Thread 0x00007fb5df3a1600]:unload_network_module, network module 0
2023-09-09 05:56:13.112459[Thread 0x00007fb5df3a1600]:on_init_network, unload_network_module, result = 0
2023-09-09 05:56:13.112463[Thread 0x00007fb5df3a1600]:preset_folder: set to empty
2023-09-09 05:56:13.112494[Thread 0x00007fb5df3a1600]:load_presets enter, substitution_rule 3, preferred printer_model_id 
2023-09-09 05:56:13.115013[Thread 0x00007fb5df3a1600]:load_vendor_configs_from_json: parse /home/mirekspanko/.config/OrcaSlicer/system/Creality.json, got description:  "Creality configurations"
2023-09-09 05:56:13.147697[Thread 0x00007fb5df3a1600]:load_vendor_configs_from_json: parse /home/mirekspanko/.config/OrcaSlicer/system/Custom.json, got description:  "My configurations"
2023-09-09 05:56:13.159945[Thread 0x00007fb5df3a1600]:update_compatibility for all presets enter
2023-09-09 05:56:13.159974[Thread 0x00007fb5df3a1600]:update_compatibility for all presets exit
2023-09-09 05:56:13.159980[Thread 0x00007fb5df3a1600]:update_compatibility for all presets enter
2023-09-09 05:56:13.159987[Thread 0x00007fb5df3a1600]:update_compatibility for all presets exit
2023-09-09 05:56:13.166130[Thread 0x00007fb5df3a1600]:update_compatibility for all presets enter
2023-09-09 05:56:13.166192[Thread 0x00007fb5df3a1600]:update_compatibility for all presets exit
2023-09-09 05:56:13.166202[Thread 0x00007fb5df3a1600]:load_presets finished, returned substitutions 0
2023-09-09 05:56:13.166214[Thread 0x00007fb5df3a1600]:create the main window
2023-09-09 05:56:13.262654[Thread 0x00007fb5df3a1600]:CreateWebView: file:/tmp/.mount_OrcaSlqIX7bS/resources/web/homepage/index.html
2023-09-09 05:56:13.501805[Thread 0x00007fb5df3a1600]:select_technology: change the printer technology from 2 to 0
2023-09-09 05:56:13.562744[Thread 0x00007fb5df3a1600]:get_current_slice_result:m_current_plate 0, current_plate 0x56245e7f2270
2023-09-09 05:56:13.586201[Thread 0x00007fb5df3a1600]:set_current_panel: current_panel 0, new_panel 0x56245e926090
2023-09-09 05:56:14.075459[Thread 0x00007fb5df3a1600]:ImageGrid::SetStatus: No printers.
2023-09-09 05:56:14.078499[Thread 0x00007fb5df3a1600]:monitor: show_status = 4
2023-09-09 05:56:14.084795[Thread 0x00007fb5df3a1600]:load_ams_list: 0 items
[... 397 lines cut ...]
2023-09-09 06:01:54.150489[Thread 0x00007fb5df3a1600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:35}
2023-09-09 06:02:04.150453[Thread 0x00007fb5467fc6c0]:process_task{ type:Backup, id:0, path:, object:0, extra:36}
2023-09-09 06:02:04.150692[Thread 0x00007fb5df3a1600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:36}
2023-09-09 06:02:14.150465[Thread 0x00007fb5467fc6c0]:process_task{ type:Backup, id:0, path:, object:0, extra:37}
2023-09-09 06:02:14.150670[Thread 0x00007fb5df3a1600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:37}
2023-09-09 06:02:24.150360[Thread 0x00007fb5467fc6c0]:process_task{ type:Backup, id:0, path:, object:0, extra:38}
2023-09-09 06:02:24.150510[Thread 0x00007fb5df3a1600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:38}
2023-09-09 06:02:34.150380[Thread 0x00007fb5467fc6c0]:process_task{ type:Backup, id:0, path:, object:0, extra:39}
2023-09-09 06:02:34.150496[Thread 0x00007fb5df3a1600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:39}
2023-09-09 06:02:44.150455[Thread 0x00007fb5467fc6c0]:process_task{ type:Backup, id:0, path:, object:0, extra:40}
2023-09-09 06:02:44.150685[Thread 0x00007fb5df3a1600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:40}
2023-09-09 06:02:54.150440[Thread 0x00007fb5467fc6c0]:process_task{ type:Backup, id:0, path:, object:0, extra:41}
2023-09-09 06:02:54.150691[Thread 0x00007fb5df3a1600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:41}
2023-09-09 06:03:04.150376[Thread 0x00007fb5467fc6c0]:process_task{ type:Backup, id:0, path:, object:0, extra:42}
2023-09-09 06:03:04.150461[Thread 0x00007fb5df3a1600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:42}
2023-09-09 06:03:14.150454[Thread 0x00007fb5467fc6c0]:process_task{ type:Backup, id:0, path:, object:0, extra:43}
2023-09-09 06:03:14.150685[Thread 0x00007fb5df3a1600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:43}
2023-09-09 06:03:24.150380[Thread 0x00007fb5467fc6c0]:process_task{ type:Backup, id:0, path:, object:0, extra:44}
2023-09-09 06:03:24.150472[Thread 0x00007fb5df3a1600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:44}
2023-09-09 06:03:34.150386[Thread 0x00007fb5467fc6c0]:process_task{ type:Backup, id:0, path:, object:0, extra:45}

### Checklist of files to include

- [ ] Log file
- [ ] Project file

mirspan, 2023-09-09:
To run the *.appimage package under Manjaro Linux 23.0 KDE it is necessary to have webkit2gtk and webkit2gtk-4.1 installed.

TexZeTech, 2023-09-12:
is it the same as https://github.com/SoftFever/OrcaSlicer/issues/1808 ?

closed 2023-12-19 by github-actions[bot]