OrcaSlicer/OrcaSlicer #2044: Orca sliced files produces weird artifact at times.

actmoon, 2023-09-09:
### OrcaSlicer Version

1.6.6

### OS version

Windows 11

### Printer

Bambu Labs P1S

### How to reproduce

Sliced with same settings on Bambu Slicer. 

No additional steps.

### Actual results

Orca SLicer's sliced files' preview shows artifacts. Which is also present when it is physically printed. Bambu Slicer's preview did not show artifacts for the same model, and settings, and results in a clean model when printed.

### Expected results

Both slicer should produce the same clean model preview & printed results.

### Project file & Debug log uploads

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2023-09-09:
Could you attach the project file?

also please check the below:
1. Are you slicing the model using Arachne?
2. Are you using a single top layer mode?

if so, could you disable single top layer and see what the sliced file looks like?

actmoon, 2023-09-09:
> Could you attach the project file?
> 
> also please check the below:
> 
> 1. Are you slicing the model using Arachne?
> 2. Are you using a single top layer mode?
> 
> if so, could you disable single top layer and see what the sliced file looks like? This may be related to #2034

Uploaded both project files from Orca & Bambu.

to answer your questions:

1. Yes already on Arachne
2. Nope 4 top layers

igiannakas, 2023-09-09:
This is user error. you have your seam gap set to 10mm -> the seam will be open for 10mm!

[image]

Change it to 10% - works great:

[image]

actmoon, 2023-09-09:
> This is user error. you have your seam gap set to 10mm -> the seam will be open for 10mm!
> 
> [image]
> 
> Change it to 10% - works great:
> 
> [image]

Ahhh!!

many thanks! Must have changed it accidentally when I was tweaking the seams.

closed 2023-09-09 by actmoon