OrcaSlicer/OrcaSlicer #2045: Show estimated print time in the ruler like on prusa

cpasoft, 2023-09-09:
It would be nice port this feature to Orca...

[image]
[image]

discip, 2023-09-10:
## Second that!

Additionally, regarding inserting things into the model mid print:

It would be really convenient if there was a line for each 'pause' in the final **_'Total Estimate - Info Board'_** so that you know when to be ready to insert what ever you planned into the print.
Currently, estimating when each break will occur is a relatively laborious process.

Eldenroot, 2023-09-28:
This is annoying, I miss this feature from PS.

@SoftFever

niklasb22, 2023-10-02:
I will have a look at It

cpasoft, 2023-10-02:
Great!

discip, 2023-10-08:
@cpasoft
Was addressed in #2305.
Please don't forget to close. 😊

cpasoft, 2023-10-08:
> @cpasoft Was addressed in #2305. Please don't forget to close. 😊

Great. Thank you so much :-)

Closing then...

closed 2023-10-08 by cpasoft