OrcaSlicer/OrcaSlicer #2046: Overhang logic isn't working as expected

R4toLoko, 2023-09-09:
### OrcaSlicer Version

1.6.6

### OS version

Linux Mint Cinnamon 23.1

### Printer

Voron Trident

### How to reproduce

SoftFever/OrcaSlicer always have hard time dealing with overhang.
Even when using a block modifier, it ignores the speeds.

[image]

While PrusaSlicer without a block modifier using Classic mode, it does respect the speed.

[image]

1. pick any print with overhang
2. slice it and observe the rules being ignored

### Actual results

As consequence, you have beautiful prints destroyed:
[image]

### Expected results

Slow down for overhang like PrusaSlicer does.
I opened a few tickets already about overhang but it doesn't seem to have any priority to fix it.
I did shared the other day having an spectacular overhanging print but that was a hit and miss.

### Project file & Debug log uploads

[landing.zip](https://github.com/SoftFever/OrcaSlicer/files/12565957/landing.zip)
[debug_Sat_Sep_09_23_26_42_4804.log](https://github.com/SoftFever/OrcaSlicer/files/12565965/debug_Sat_Sep_09_23_26_42_4804.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2023-09-09:
Hi,

I don't quite understand the problem here.
I see you set all speed to 60 with a block modifier, the area is indeed 60 speed as shown in the screen shot?
Did I miss anything?
[image]

If you want the overhang to slowdown, you can try following setting
[image]

R4toLoko, 2023-09-09:
Thank you for the support.

I'm using Voron cube as test, it's faster to print.

- Left: PrusaSlicer
- Right: OrcaSlicer
- same silk filament
- same 80mm/s printing speed
- same 100% cooling
- same classic overhang speed set to 60mm/s
- middle of the bed sheet

Noticed how the X has sharp corner and a very tiny imperfection at the top.
With OrcaSlicer, noticed how the sharp corner is gone and a little imperfection at the top.

Y side, it almost made while with OrcaSlicer didn't performed as well.
I forgot to check the seam orientation, I'll redo the test tomorrow.

This is the problem I'm having across my prints since the start. It gets worse with soft filament.

Overhang support is a hit and miss. I'm already printing slow compared with my normal 200mm/s 5k accel and below 50mm/s generates too much resonance so why 60.
Will do another test tomorrow morning.

[image]

R4toLoko, 2023-09-10:
Also, I have noticed this behaviour.
The very same model...

[image]

... sliced for many prints have one overhang logic.
Knowing that the speed is 80mm/s, it is not slowing down enough.

[image]

If I slice the same model for a single print, look how different the logic gets.
Also, it is printing way out of the chart, I have no speeds below 60mm/s.

My printer generates an insane resonance at 50mm/s so I never go below 60mm/s and yet OrcaSlicer is trying to print even at 12mm/s, that is insane.

[image]

If I try that with PrusaSlicer, I get what you would expect:

[image]

R4toLoko, 2023-09-10:
I might have found why the single print has such weird logic:
**Min print speed: 10mm/s**

[image]

Setting the min speed to 60mm/s, it kind of stops trying to print crazy slow but the whole logic gets broken locked at 60mm/s

[image]

I am recreating the same profiles on PrusaSlicer trying to understand why they behave so differently for overhangs.

igiannakas, 2023-09-10:
I've been doing some digging on this @SoftFever 

Firstly it appears that indeed there is a variance in the logic even though the algorithm between Prusa slicer and Orca slicer is similar/same.

The thing is I am not sure which one of the two slicers has a bug!

Example - Benchy, sliced at 0.16 layer height on both slicers - Orca using the latest Dev branch committed a few hours ago, containing commit 0b3dfa3.
[Test files.zip](https://github.com/SoftFever/OrcaSlicer/files/12567892/Test.files.zip)

In the tip of the benchy we have an overhang. **From close inspection of the layer overlap it appears to be around 75% or so.**

**Orca slicer:**
[image]

**Prusa slicer:**
[image]

**Key differences:**
**1. Prusa slicer slows down to the lowest speed (20mm/sec) on the tip of the bow even though the overlap is 75% or their abouts.** This seems incorrect on Prusa's side as clearly the overlap is more than 50-75% or so. Changing the overlap speeds to be 100% for all but the 0% overlap still results in a slow down to 26mm/sec (from 40mm/sec external perimeter speed), which means that the slicer considers the bow to be a significant overhang where it clearly isn't. Unless there is some further logic somewhere that influences this. See below:
[image]

**2. Prusa slicer appears to be slowing down the internal perimeters where an external overhang perimeter is detected**. It is obvious from the above as Orca doesnt slow down the internal perimeters vs. Prusa slowing them down at the tip of the bow. I am not sure whether this is beneficial or where in the code that slowdown is implemented on Prusa but may be worth doing some digging on that.
 
3. Prusa slicer considers an overhang wall an overhang and not a bridge - which means it doesnt change the speed mid extrusion vs. Orca slicer that does. See below:

**Prusa slicer -> Speed remains constant over the overhang and is sourced from the dynamic overhang speed configuration**
[image]

**Orca slicer -> Speed is set to the bridge speed for an overhang wall and the overhang speed is ignored for this**
[image]

**So in summary, it appears to:**
**1. Prusa slicer may have a bug that causes the perimeters to be aggressively slowed** down even there is enough overlap in order to print them correctly. Or there is an undocumented feature that influences that slow down on the bow of the benchy. Mind you, this may not be a "bad" bug as slowing down aggressively there is a benefit - Orca does cause slightly worse overhangs even though technically the setting is working correctly.
**2. Orca slicer may have a bug where the bridge speed is used for an overhang wall.**

I believe bug (2) is caused by the following code segment:

```
             FloatOrPercent{m_config.get_abs_value("bridge_speed") * 100 / out_wall_ref_speed, true},
              FloatOrPercent{m_config.get_abs_value("bridge_speed") * 100 / out_wall_ref_speed, true}})
```
Line 4548 in the GCode.cpp file. I think the above should be set to the lower of the bridge speed or overhang speed for the 75-100% overlap bracket as we dont have a dedicated 0% overlap option as Prusa slicer does.

**3. There is a difference in handling internal perimeters neighbouring overhanging walls.** Here I am not sure what is the best approach. To me it appears that Prusa slicer is being overly conservative to be reducing internal wall speed when the external perimeter is an overhang, but that may have quality benefits that I can't see.

R4toLoko, 2023-09-10:
Thank you so much for sharing your experience  @igiannakas 🙏

I've been trying to deal with this for months but I always fail to provide as much information as possible.

I don't use support so I do depend a lot on the printer cooling and overhang logic.

On PrusaSlicer like OrcaSlicer, I increased the min speed that was 10mm/s and 20mm/s respectively to 60mm/s and the classic Overhang speed set to 60mm/s, PrusaSlicer stopped printing too slow and more like what you would expect.
On OrcaSlicer this option is within the printer tab and PrusaSlicer is within the filament setting IIRC.

I cannot replicate the same with OrcaSlicer, it either prints too fast or ignore the speed altogether.
It's frustrating printing gorgeous models only to notice the overhang looking very bad 🤕

For now, I'm having to print overhang prints with PrusaSlicer, after the changes it's printing better overhang but not perfect yet as haven't used it for ages so I'm changing as I go.

Thank you again for the input, hopefully we can get OrcaSlicer to print better overhang 🙏

igiannakas, 2023-09-10:
@SoftFever 

Think I found the code that causes this behaviour in Prusa slicer: **"Prusa slicer slows down to the lowest speed (20mm/sec) on the tip of the bow even though the overlap is 75% or their abouts"**

```
            // The following code artifically increases the distance to provide slowdown for extrusions that are over curled lines
            float artificial_distance_to_curled_lines = 0.0;
            const double dist_limit = 10.0 * path.width;
            {
                Vec2d middle = 0.5 * (curr.position + next.position);
                auto line_indices = prev_curled_extrusions[current_object].all_lines_in_radius(Point::new_scale(middle), scale_(dist_limit));
                if (!line_indices.empty()) {
                    double len   = (next.position - curr.position).norm();
                    // For long lines, there is a problem with the additional slowdown. If by accident, there is small curled line near the middle of this long line
                    //  The whole segment gets slower unnecesarily. For these long lines, we do additional check whether it is worth slowing down.
                    // NOTE that this is still quite rough approximation, e.g. we are still checking lines only near the middle point
                    // TODO maybe split the lines into smaller segments before running this alg? but can be demanding, and GCode will be huge
                    if (len > 8) {
                        Vec2d dir   = Vec2d(next.position - curr.position) / len;
                        Vec2d right = Vec2d(-dir.y(), dir.x());

                        Polygon box_of_influence = {
                            scaled(Vec2d(curr.position + right * dist_limit)),
                            scaled(Vec2d(next.position + right * dist_limit)),
                            scaled(Vec2d(next.position - right * dist_limit)),
                            scaled(Vec2d(curr.position - right * dist_limit)),
                        };

                        double projected_lengths_sum = 0;
                        for (size_t idx : line_indices) {
                            const CurledLine &line   = prev_curled_extrusions[current_object].get_line(idx);
                            Lines             inside = intersection_ln({{line.a, line.b}}, {box_of_influence});
                            if (inside.empty())
                                continue;
                            double projected_length = abs(dir.dot(unscaled(Vec2d((inside.back().b - inside.back().a).cast<double>()))));
                            projected_lengths_sum += projected_length;
                        }
                        if (projected_lengths_sum < 0.4 * len) {
                            line_indices.clear();
                        }
                    }

                    for (size_t idx : line_indices) {
                        const CurledLine &line                 = prev_curled_extrusions[current_object].get_line(idx);
                        float             distance_from_curled = unscaled(line_alg::distance_to(line, Point::new_scale(middle)));
                        float             dist                 = path.width * (1.0 - (distance_from_curled / dist_limit)) *
                                     (1.0 - (distance_from_curled / dist_limit)) *
                                     (line.curled_height / (path.height * 10.0f)); // max_curled_height_factor from SupportSpotGenerator
                        artificial_distance_to_curled_lines = std::max(artificial_distance_to_curled_lines, dist);
                    }
                }
```

This code seems to evaluate how close the middle of a line segment (represented by curr.position and next.position) is to previous curled extrusions. If the line segment is close enough to any curled extrusion, it computes a measure (artificial_distance_to_curled_lines) of this proximity. This then is used below:
`float curled_base_speed = interpolate_speed(speed_sections, artificial_distance_to_curled_lines);`
and
`float final_speed       = std::min(curled_base_speed, extrusion_speed);`

To take the minimum speed between the curled line interpolation or the interpolated overhang speed.

IMHO this is a great improvement as it proactively takes into account possible edge curling and reducing the speed even though the extrusion is not overhanging much.

I'll have a go at implementing this but my coding skills are a bit rusty so I may fail miserably in doing it. Would love if you could take a look at the above and lets try a proof of concept.

igiannakas, 2023-09-11:
Proof of concept for slowing down for perimeters similarly to Prusa Slicer is done - tweaked for HS printing as Prusa slicer was too aggressive I found with the slow downs.

[image]

Now on to debugging it and potentially making that a UI option so its selectable by the user.

R4toLoko, 2023-09-12:
@igiannakas thank you so much for doing the lord's work.
I don't know to code but I am happy to help with testing.
I saw your comment about the deformation on the hull and one of the reasons I stopped printing it, mine looks a completely disaster specially coz I am printing at 200mm/s and 5k accel. 
It looks almost like if it did hit another sip at the port haha

This is connected with this ticket: https://github.com/SoftFever/OrcaSlicer/issues/44
Once we get them in place, man, the printing quality is gonna be wild.

Thank you for your time on this.

igiannakas, 2023-09-12:
> @igiannakas thank you so much for doing the lord's work. I don't know to code but I am happy to help with testing. I saw your comment about the deformation on the hull and one of the reasons I [...]

That is actually the next thing I’d like to port - pressure equalizer. I used this feature constantly on prusa slicer and am missing it on my Bambu printer now with orca. Indeed these two things make a big difference in print quality.

The PR above has been tagged for build by softfever, hopefully it will be available soon in the nightly builds for testing.

igiannakas, 2023-09-12:
Ive started working on it here: https://github.com/igiannakas/OrcaSlicer/tree/Pressure-equalizer

There is quite a bit of the code already present since the original Bambu studio fork was taken, but it's out of date or slightly different compared to Prusa slicer 2.6.1. I've ported the latest pressure equaliser code over but have not enabled it yet. However it compiles successfully after I've retrofitted it to the older Bambu Studio/Orca slicer data. In summary around 50-70% of the work is already done on the above fork. I'll carry on working on this time allowing and hopefully will be able to have a working code published as a PR here in a few weeks.

igiannakas, 2023-09-12:
Developer build for the overhang changes is here: https://github.com/SoftFever/OrcaSlicer/actions/runs/6159766903

feedback appreciated!

R4toLoko, 2023-09-12:
@igiannakas, there is a high chance, 99.99%, of me being a dumbass but I will ask anyway
This is your build

[image]

And this is the current 1.7.0-beta, they are virtually the same.
Am I missing something? haha
I don't have classic mode enabled, only the slicer logic itself.
Thanks man

[image]

R4toLoko, 2023-09-12:
Also, your print has a completely different behavior.
Yours keep the same speed across the model while mine have the speeds all over the place.
This is one thing I always tried to understand why and how to remove it to only really slow down when needed.
I found this option that for some weird reason was set to 10mm/s but that isn't it either.

[image]

R4toLoko, 2023-09-13:
I am lost to explain this @igiannakas @SoftFever 

I am printing this 8h print with PrusaSlicer
OrcaSlicer on the left and PrusaSlicer on the right.
Same 200mm/s and 5k acceleration but notice how the speed is completely different.

[image]

Now, here is where things get weird.
Nothing changed after enabling the new feature.

[image]

If I disable this on PrusaSlicer, the logic breaks

[image]

[image]

Now, if I set both PS and OS like so:

[image]
[image]

Both slicers have virtually the same behaviour .... virtually!

[image]

PS still having better speed control while OS breaks down.

[image]

If I disable slope negative on PrusaSlicer, nothing really change much so that is not it.

[image]

So far, it has been printing a very nice overhang, this isn't a hard overhang but there is zero issues.
idk how to explain this haha

[image]

[image]

igiannakas, 2023-09-13:
> I am lost to explain this @igiannakas @SoftFever
> 
> I am printing this 8h print with PrusaSlicer OrcaSlicer on the left and PrusaSlicer on the right. Same 200mm/s and 5k acceleration but notice how the speed is completely different.
> 
> [image]
> 
> Now, here is where things get weird. Nothing changed after enabling the new feature.
> 
> [image]
> 
> If I disable this on PrusaSlicer, the logic breaks
> 
> [image]
> 
> [image]
> 
> Now, if I set both PS and OS like so:
> 
> [image] [image]
> 
> Both slicers have virtually the same behaviour .... virtually!
> 
> [image]
> 
> PS still having better speed control while OS breaks down.
> 
> [image]
> 
> If I disable slope negative on PrusaSlicer, nothing really change much so that is not it.
> 
> [image]
> 
> So far, it has been printing a very nice overhang, this isn't a hard overhang but there is zero issues. idk how to explain this haha
> 
> This model is being printed with the speeds above with PS.
> 
> [image]
> 
> [image]

Can you upload both model files so I can take a look?

in terms of how to match the toggles between orca and prusa:
1. Prusa doesn’t give you an option to disable the detection of curled edges. 
2. If you want to compare like for like the slowdown for overhang and curled edges need to be on in orca to match the feature on prusa. 

Code wise there is very little difference between the two slicers now in that area. The differences are:
1. Orca doesn’t have pressure equalizer so always disable that in prusa to compare like for like
2. Orca uses the bridging speed when detection of curled edges is off for any perimeter with an overlap less than 13%. When curled edges are on, they are identical - they both use the overhang speeds 
3. For the benchy, I disabled slowdown for layer cooling in the filament as I wanted to see the effect of the slowdown for curled edges without the filament slowdown interfering. Try the same and it will match 

Zip the model and submit so I can have a like for like look with what you’re seeing and we can get to the bottom of it.

R4toLoko, 2023-09-13:
Curled edges is on and slope negative is disabled, there are no changes.
Both slicer bridge speed is set to 100mm/s
I am sending you this engine attached and also the project so you will see what I am seeing.

Thanks man

[engine.zip](https://github.com/SoftFever/OrcaSlicer/files/12594315/engine.zip)
[project.zip](https://github.com/SoftFever/OrcaSlicer/files/12594329/project.zip)

[image]

igiannakas, 2023-09-13:
can you also attach the Prusa project file too please?

igiannakas, 2023-09-13:
A few initial observations on the Orca project attached - it seems to me like it is working as intended:

This is a cross section and close up of layer 273 - 

[image]

The full speed lines are indeed overlapping more than 90% against the extrusion below:

[image]

Then when they are overlapping a little bit less, the perimeter is slowed down:
[image]

Also the curled perimeters is also working but your settings are not allowing it to work - the speed is set to a binary 200mm/sec or 100mm/sec. For example if you set the overhang speed for the 75-100% bracket down to 20mm/sec it shows the below:

Without curled edges
[image]

With curled edges detection:
[image]

What the difference between Prusa and Orca may be and here I am speculating, is that Prusa interpolates the speed between the 0 overhang and 10% overhang of a line. That should really not be the case as with more than 90% overlap you can go full speed in terms of printing.

In Orca the speed adjustments start at the 10% overhang and more.

igiannakas, 2023-09-13:
For context this is the same model sliced with my Bambu Lab tuned preset:

[image]

It slows down where it needs to and doesnt where it doesnt :)

R4toLoko, 2023-09-13:
Yep, it all makes perfectly sense.
I don't know why my dead brain didn't see that the speed for high overhang was too high when it should be your recommendation instead 🤦

Still unclear how PrusaSlicer with the same 100mm/s for everything slowed down, I guess that's why I didn't took that into consideration, well, out of scope.

Thank you so much for checking this out. 
Will have a play and we should be good to close this ticket.

Thanks again.

igiannakas, 2023-09-13:
So, I've made a change to the Orca algorithm to start interpolating the speed from 100% overlap to the 25% overlap in a similar way that Prusa slicer does.

[image]

The results mirror each other now fully.

However I am not sure whether this is desired behaviour. Orca slicer sets a boundary to 10% overhang being the threshold where speed adjustments start. So I am not sure whether we want to override that as I believe making speed adjustments there is unnecessary.

@SoftFever the code that addresses the user feedback above is below. I'll leave it up to you on whether this is a needed change or not.

```
            auto calculate_speed = [&speed_sections, &original_speed](float distance) {
                float final_speed;
                if (distance <= speed_sections.front().first) {
                    final_speed = original_speed;
                    if (distance >0){
                    	float t = distance / speed_sections.front().first;
                    		  t           = std::clamp(t, 0.0f, 1.0f);
                    		final_speed = (1.0f - t) * original_speed + t * speed_sections.front().second;
                    	printf("distance %f, speed_sections.front().first %f, original speed %f, t %f, final_speed %f \n", distance, speed_sections.front().first, original_speed, final_speed);
                    }
                } else if (distance >= speed_sections.back().first) {
                    final_speed = speed_sections.back().second;
                } else {
                    size_t section_idx = 0;
                    while (distance > speed_sections[section_idx + 1].first) {
                        section_idx++;
                    }
                    float t = (distance - speed_sections[section_idx].first) /
                              (speed_sections[section_idx + 1].first - speed_sections[section_idx].first);
                    t           = std::clamp(t, 0.0f, 1.0f);
                    final_speed = (1.0f - t) * speed_sections[section_idx].second + t * speed_sections[section_idx + 1].second;
                }
                return final_speed;
            };
```

R4toLoko, 2023-09-13:
@igiannakas thanks for looking into that, it was an observation I noticed when tracking this down, you guys know better than me the correct logic 😅

Thanks again for this beautiful contribution 🙏🏆

igiannakas, 2023-09-13:
> Yep, it all makes perfectly sense. I don't know why my dead brain didn't see that the speed for high overhang was too high when it should be your recommendation instead 🤦
>
> Still unclear how [...]

Take a look at my comment above - I've mirrored the behaviour but there is a difference in design between Prusa and Orca in that area. Orca starts speed adjustments from 10% overhang, while Prusa from 0% overhang. I think the Orca way is better as small variations in overhang that happen all the time with imperfect models due to STL facets etc won't result in unnecessary speed changes in printing. So I'd be more inclined to be asking Prusa to set a lower limit upon which the overlap is no longer taken into account and the full speed is used.

R4toLoko, 2023-09-13:
> Take a look at my comment above - I've mirrored the behaviour but there is a difference in design between Prusa and Orca in that area. Orca starts speed adjustments from 10% overhang, while Prusa [...]

Yup, understood it crystal clear.
Keep the logic that you guys believe is the correct.
I'll starting using your build until 1.7.0 is finally released, hopefully with this change in place 😅

R4toLoko, 2023-09-15:
@igiannakas @SoftFever ,

I am closing this ticket.
Keep in mind that I am still using the very first 1.7.0-beta nightly release with this change you shared before.
I saw that 1.7.0 was released but I will wait for 1.7.1 to hopefully have this implemented.

Last update from me, this is Polymaker PolyTerra Matte PLA, it is soft like butter so a lot easier to see if things are going as expected with the printer cooling/overhang.
How I prayed for this day lmao lol

[image]

The few changes I have in place.
My printer generates too much resonance at low speed, it hates that.
This is the same 200mm/s 5k accel profile

1. Only 75-100% overhang has been slowed down

[image]

2. Min print speed increased, 20mm/s is too slow

[image]

3. Increased wall loops from 2 to 3 allows more time for cooling, otherwise, it is a bad idea for the speed I am used to print.
4. I have been using Classic wall generator because Arachne leaves voids everywhere.
Switching back to Arachne, it further helps the adventure above by removing weird patterns and added to the extra wall like mentioned above, makes everything perfect.

closed 2023-09-15 by R4toLoko