OrcaSlicer/OrcaSlicer #2050: Support Multiple Extruders

goldid, 2023-09-09:
**Is your feature request related to a problem? Please describe.**
The software does not have support more than a single extruder at the moment.

**Which printers will be beneficial to this feature**
Tenlog 3D Pro

**Describe the solution you'd like**
I'd need support for at least two extruders.

**Describe alternatives you've considered**
Using a single extruder, but that is limiting.

ultraSsak, 2023-09-10:
+1 for Prusa's MMU3

freeformz, 2023-11-06:
+1 especially since the 2 and 5 tool head Prusa XL is shipping now.

Felix14-v2, 2024-01-26:
Bump (cuz there's a stupid bot)

d-portero, 2024-03-03:
+1 as more multi tool printers are becoming available, not to mention existing IDEX printers

Klober81, 2024-03-07:
+1 I have 3x Bambu X1Cs and have been using OrcaSlicer since V1.0.10. Hands down the best slicer out there IMO. But I just got the Vivedino Marathon from Formbot which is IDEX and I'm REALLY missing OrcaSlicer. Multi-extruder setups need the Orca love too. Please save us SoftFever, Noisyfox, igiannakas and crew! <3

doobaloo132, 2024-03-08:
+1. I made an account just to ask this specific question. I'm about the pull the trigger on a Marathon and Orca is my go to for any slicer related stuff. Best one out there in my opinion.

Hardy786, 2024-03-19:
+1 for Voron Stealth Changer

vgdh, 2024-03-19:
It's not clear for me why it was cut out because it was in prusa code base.

FOTNMC, 2024-03-20:
+one from me to

gfacek, 2024-03-22:
+1

I just switched from Cura and wanted to slice for my IDEX. Also I'm building a Voron right now and intending to implement Stealth Changer.

peterek0000, 2024-03-28:
+1 for Voron toolchangers (Daksh, Stealhchanger, Tapchanger...)

adbizau, 2024-03-31:
+1 voron 2.4

chujian, 2024-04-07:
+1 for Voron toolchangers

MichaelMichaelMichaelMichaelMichael, 2024-04-24:
+1
Voron v2.4

BlackNet, 2024-04-24:
Voron 2.4, Triden plus others.  Many can support multiple tool heads easily.

zane797, 2024-04-24:
+1 for the Voron community, also with the SV08 with more extensive modding.

psychotik2k3, 2024-04-25:
+1 i'm quite surprised that orca only handle one extruder. this is blocking for me as i got my tool changer with 3 tools, each with different speeds settings etc.

psychotik2k3, 2024-04-25:
> +1 for Prusa's MMU3

this has nothing to do with MMU as MMU is one extruder and is already supported by orca as SEMM

NoeliaG2018, 2024-04-26:
+1 for Hellbot Hidra

bte2, 2024-04-29:
+1
Voron 2.4 stealthchanger

Contomo, 2024-05-25:
defintly needed are more prominent right now then ever

nic335, 2024-05-27:
+1 Voron 2.4 stealth changer

b1tfan, 2024-05-29:
+1 StealthChanger

FlisherOfatale, 2024-05-29:
+1 Would be definetly be interesting for Toolchanger, and there is probably not much missing for a basic version (ie: all toolhead with same nozzle diameter...)

rwlodkowski, 2024-06-02:
+1 Also for all Snapmakers which comest with dual-extrusion module.

PipoPshots, 2024-06-03:
+ MMU 3 Please !!!!

KilimSerg, 2024-06-15:
Hello
I would be glad to add support for several extruders 
to Orco Slicer.
Specifically in my case, I often print with PETG plastic 
and support I print with PLA plastic.
PETG and PLA sinter very poorly, which allows
print with PETG plastic on PLA support without any intermediate gaps.
After printing, it is very easy to separate the supports without 
compromising quality surface printed with PETG plastic.

broncosis, 2024-06-23:
@SoftFever any progress with this I am happy to test beta or alpha builds

freeformz, 2024-06-24:
FWIW: I am a monthly sponsor via Kofi ... If you use OrcaSlicer extensively and can afford as little as $5/mo may I humbly suggest you do the same: https://ko-fi.com/G2G5IP3CP

xanderphillips, 2024-06-27:
+1 Voron 2.4 stealth changer

SarraceniaWilds, 2024-06-30:
Need custom tool change gcode field, multiple extruder settings, not sure how anyones been setting an x y offset for second extruder using orca as it stands. Big headache.

broncosis, 2024-06-30:
For the tool changers the xy offset is in the firmware 

⁣Get BlueMail for Android ​

SoftFever, 2024-06-30:
thanks all.
This duplicated with https://github.com/SoftFever/OrcaSlicer/discussions/5942
I suggest we have all the multi tool discussion in one place

SoftFever, 2024-07-10:
Hey guys,

I've added support for tool changer printers. I'm seeking feedback and testers. If you have experience with TapChanger, StealthChanger, or similar systems, are skilled in analyzing G-code, and are up for a challenge, please consider helping me test the build here: https://github.com/SoftFever/OrcaSlicer/pull/6087

NEVO3D, 2024-07-20:
> Hey guys,
> 
> I've added support for tool changer printers. I'm seeking feedback and testers. If you have experience with TapChanger, StealthChanger, or similar systems, are skilled in analyzing G-code, and are up for a challenge, please consider helping me test the build here: #6087
> 
> ~⚠️WARNING⚠️ : Please proceed with caution if you decide to try this build. I **haven't tested it on an actual printer yet**, though the G-code looks fine so far. I'm seeking more experienced users to help inspect the generated G-code.~
> 
> 🚀GOOD NEWS🚀: Multiple successful prints have been reported; it should be safe to try [Discord Thread](https://discord.com/channels/1137181739773603922/1258993224018821160). **For bugs, please kindly help by logging issues with details and sample projects, then label them as ToolChanger. This will help me track them.**

I

Lungenpest, 2024-07-23:
+1 Prusa XL 5T

Felix14-v2, 2024-10-22:
Resolved, I guess?

freeformz, 2024-10-23:
Closable now AFAICT

Felix14-v2, 2025-01-22:
?

closed 2025-04-30 by github-actions[bot]