OrcaSlicer/OrcaSlicer #2062: Organic support gets confused by adaptive layer height

szabczy, 2023-09-10:
### OrcaSlicer Version

1.7 Beta

### OS version

Windows 10

### Additional system information

CPU: 11th Gen Intel(R) Core(TM) i5-11600 
RAM: 32GB
GPU: RTX 3060

### Printer

Creality Ender 5

### How to reproduce

1. Enable Adaptive layer height
2. Support type Tree(auto), style Organic
3. Slice plate

### Actual results

[image]
[image]
[image]

### Expected results

The organic support should support the problematic areas of the model.

### Project file & Debug log uploads

No project file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jeremytodd1, 2023-09-10:
Just to get this out there, PrusaSlicer's Organic supports also do not work with adaptive layer height.

[image]

szabczy, 2023-09-10:
Then I guess it's not a bug.
It would be nice to have, though.

closed 2023-12-18 by github-actions[bot]