OrcaSlicer/OrcaSlicer #2063: Bulging Seams: Innermost wall unretract/deretract amount dependent on Sparse Infill Line Width

mattsiri, 2023-09-10:
### OrcaSlicer Version

1.7.0-beta

### OS version

Windows 11

### Printer

Bambu X1 Carbon

### How to reproduce

1. Choose a model
2. Choose any filament profile (Bambu PLA Basic for simplicity).
3. Choose any process profile (0.20mm Standard @BBL X1C for simplicity).
4. Set wall loops to 2 or more.
6. Slice.
7. Show Retract and Unretract in Preview (Color Scheme: Line Type).
8. Scroll down past top layers.
9. Look at Unretract blue dot.
[image]
11. Set Sparse Infill Line width to 0.7mm
12. Reslice and compare Unretract blue dot
[image]

### Actual results

The unretract amount of the innermost wall seam varies with the Line Width of the Sparse Infill. Printing with 0.7mm Sparse Infill Line width results in bulging seams.

### Expected results

The unretract amount of the innermost wall seam should be independent on the Line Width of the Sparse Infill - it should only vary with Inner Wall Line Width.

### Project file & Debug log uploads

[UnretractBug_InnermostWall_SparseInfill - 2023-09-10.zip](https://github.com/SoftFever/OrcaSlicer/files/12569000/UnretractBug_InnermostWall_SparseInfill.-.2023-09-10.zip)
[Logs.zip](https://github.com/SoftFever/OrcaSlicer/files/12569001/Logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

mattsiri, 2023-09-10:
Note: This issue was present in V1.6.6 as well. I hadn't gotten a chance to report it until today.

igiannakas, 2023-09-10:
Is this just a visual glitch or is it in the gcode too? You should be able to see whether the E amount is larger in the former.

gennro, 2023-09-10:
I am noticing this same issue during printing on 1.6.6 as well.

mattsiri, 2023-09-10:
Good call. I checked. It's a bit odd.
It looks to be strictly visual with default retraction settings.

For Bambu PLA Basic, 0.20mm Standard @BBL X1C:
- SparseWidth = 0.45mm : G1 E.8 F1800
- SparseWidth = 1mm : G1 E.8 F1800

However, there is another related issue when using Extra Length on Restart. This setting is not applied in some cases unless Avoid Crossing Walls is enabled. I'll post screenshots, files, and test print pics tomorrow.

For context, I print a lot of polypropylene. It looks like PP might need separate PA values for seam, corner, and top. For now, however, using a small amount of negative Extra Length on Restart helps quite a bit with seam bulge as the PA to keep edges clean is lower than the PA that cleans up the seams.

mattsiri, 2023-09-11:
The following examples show that the Unretract blue dot is a visual bug and that the extra length on restart is not being applied as expected.

### Test 1
1. Add a cylinder primitive.
2. Profiles: [Bambu Lab X1 Carbon 0.4 nozzle, Textured PEI Plate, Bambu PLA Basic @BBL X1C, 0.20mm Standard @BBL X1C]
3. Change Sparse Infill speed to 100mm/s (in case of testing with extra wide sparse infill).
4. In the filament profile, enable Extra length on restart override, set to -0.15mm.
5. Slice.
6. View a non-top or non-bottom layer.

Results:
[image]
The extra length on restart is not applied. The default retraction length is 0.8mm and the resulting deretraction length is 0.8mm. The expected deretraction length is 0.65mm. Changing the extra length on restart to a positive value has no impact.

### Test 2
1. Complete Test 1.
2. Change the Sparse Infill Line Width to 1mm.
3. Reslice.

Results:
[image]
Again, the extra length on restart is not applied. However, the blue dot representing the Unretract command is significantly larger. Adjusting the Sparse Infill Line Width up and down demonstrates that the size of the blue dot on the inner wall is dependent upon the Sparse Infill Line Width.

Looking at the Unretract move at the start of the sparse infill again shows a length of 0.8mm (extra length still not applied). However, the blue dot diameter is smaller than expected.
[image]

Setting the Outer Wall Line Width to 0.7mm shows that the blue dot diameter at the beginning of the sparse infill increases. It is still not as large as the blue dot at the beginning of the inner wall. 
[image]

Test 2 suggests that the diameter of the blue dot is dependent on the width of the line last printed rather than the deretraction length. Both tests demonstrate that the extra length on restart is not applied.

Files for Tests 1 and 2:
[Deretraction Bug - Tests 1 and 2 - 2023-09-11.zip](https://github.com/SoftFever/OrcaSlicer/files/12577120/Deretraction.Bug.-.Tests.1.and.2.-.2023-09-11.zip)

mattsiri, 2023-09-11:
### Test 4a
1. Using the profiles from Test 1, add a 3DBenchy.
2. Slice and look at layer 56.

Results:
Looking at the Unretract lines in the gcode for layer 6 shows the following:
_36113 G1 E.8 F1800
36167 G1 E.8 F1800
36213 G1 E.8 F1800
36412 G1 E.8 F1800
36498 G1 E.8 F1800
36519 G1 E.8 F1800
36534 G1 E.8 F1800
36869 G1 E.8 F1800
36888 G1 E.65 F1800
36989 G1 E.8 F1800
37008 G1 E.8 F1800
37096 G1 E.8 F1800_

What is going on with line 36888?
[image]

The extra length on restart is applied properly after a sparse infill set and before an internal solid infill set. This is not always the case (line 41933).

### Test 4b
1. (as Test 4a)
2. change Sparse Infill Line Width from 0.45mm to 1mm.
3. View a similar deretraction event as line 36888 in Test 4a. (line 34407)

Results:
[image]

Deretraction length is back to 0.8mm.

Files for Test 4:
[Deretraction Bug - Test 4 - 2023-09-11.zip](https://github.com/SoftFever/OrcaSlicer/files/12577745/Deretraction.Bug.-.Test.4.-.2023-09-11.zip)

mattsiri, 2023-09-11:
Disabling wipe while retracting has an impact...

Test 5
1. Start with .3mf file from Test 4 (same profiles as Test 1).
2. In the filament profile, override the Wipe while retracting setting and set it to false.
3. Slice and look retraction commands for layer 56.

Results:
[image]

The first deretraction is 0.65mm as expected. Here are the deretractions on layer 56:
35007 G1 E.65 F1800
35057 G1 E.65 F1800
35099 G1 E.65 F1800
35295 G1 E.65 F1800
35378 G1 E.65 F1800
35406 G1 E.65 F1800
35738 G1 E.65 F1800
35752 G1 E.65 F1800
35850 G1 E.65 F1800
35866 G1 E.65 F1800
36951 G1 E.65 F1800

Even setting the Sparse Infill line width to 1mm does not mess up the deretractions. It would seem that the deretraction issue has something to do with the Wipe While Retracting setting.

Files for Test 5:
[Deretraction Bug - Test 5 - 2023-09-11.zip](https://github.com/SoftFever/OrcaSlicer/files/12579258/Deretraction.Bug.-.Test.5.-.2023-09-11.zip)

mattsiri, 2023-09-11:
Edit: Changed title from "Bulging Seams: Innermost wall unretract/deretract amount dependent on Sparse Infill Line Width" to Deretraction Issues: Extra Length on Restart sometimes ignored and Unretract graphic diameter tied to previous line width.

closed 2023-12-19 by github-actions[bot]