OrcaSlicer/OrcaSlicer #2067: Invalid speed in 'G1 F-2147483648' when printing purge tower

gennro, 2023-09-10:
### OrcaSlicer Version

1.7.0-beta

### OS version

Win11

### Additional system information

7950x, nvidia 3090, 128gb of ram.

### Printer

Voron Trident ERCF

### How to reproduce

Any print with a purge tower with latest release.

### Actual results

Klipper errors out with Invalid speed in 'G1 F-2147483648'

HEIGHT:0.2
;TYPE:Prime tower
;WIDTH:0.5
;--------------------
; CP TOOLCHANGE START
; toolchange #1
; material : ABS -> ABS
;--------------------
M220 S100
; CP TOOLCHANGE UNLOAD
;WIDTH:0.6
; Ramming
;WIDTH:0.5
; Retract(unload)
M104 S270
; Cooling
G1  X209.669 Y140.463 
G1 F3000
G1 F-2147483648
G1 F3000
G1 F-2147483648
G1 F3000
G1 F-2147483648
G1 F3000
G1 F-2147483648
; Cooling park
G1 F2000
G4 S0
; filament end gcode 

### Expected results

Printer should continue printing.

### Project file & Debug log uploads

[klippy(2).zip](https://github.com/SoftFever/OrcaSlicer/files/12570614/klippy.2.zip)
[debug_Sun_Sep_10_12_53_29_34780.log.zip](https://github.com/SoftFever/OrcaSlicer/files/12570617/debug_Sun_Sep_10_12_53_29_34780.log.zip)
[1_10_scale_traffic_drum_base_blk.zip](https://github.com/SoftFever/OrcaSlicer/files/12570618/1_10_scale_traffic_drum_base_blk.zip)
[1_10_scale_traffic_drum_base_blk_ABS_36m45s.zip](https://github.com/SoftFever/OrcaSlicer/files/12570624/1_10_scale_traffic_drum_base_blk_ABS_36m45s.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2023-09-11:
@gennro 
Thanks for reporting the issue.
The issue is becuase the application didn't handle it well when "Cooling tube position" is 0.
I have a fix ready but before I push the fix I want to double if this is intended behavior.
I saw you have disabled ramming and all those cooling parameters, in that case, are you handling these actions in Klipper macro? 

Is following gcode are desired result in the case?
[image]

gennro, 2023-09-11:
Correct, everything is done in macros. This is how I use prusa slicer. All I need the slicer to do is the purge tower.

thisiscam, 2023-09-11:
I think in Prusaslicer we had an option to disable tip forming? Can we have that? The screenshot above shows "; Cooling park" gcode which seems to be moving the toolhead to somewhere --- could we not have that?

closed 2023-11-19 by SoftFever

tinyfluffs, 2024-02-02:
Worth mentioning for future reference, this is a bug I'm experiencing in PrusaSlicer 2.7 as well- but thank you for fixing it here!

TMTYD, 2024-07-21:
Hi, I also got the error, what is the solution so that the error doesn't appear again. I use Klipper and prusaslicer

SoftFever, 2025-04-09:
> Exact same issue still occurring in Version 2.9.1, This is neither fixed or done.
> 
> there is no way to disable the cooling action, and if tube lenght is set to 0 it generates the G1 F-2147483648 code for every tool change.
> 
> Current WORKAROUND:
> 
> Set tube length and position settings to an impossibly LOW number (i used 0.000000001) and this will generate moves of 0 in the gcode. While this is less than ideal since it sill adds wait time it at least works around the error till it can actually be addressed with a disable button in the Single Extruder Multi Material Setup Settings Page for the Printer settings.

Can you upload a project file?