OrcaSlicer/OrcaSlicer #2069: Wrong first layer coordinates when using pressure advance calibration (lines and pattern)

jornamon, 2023-09-11:
### OrcaSlicer Version

1.7.0-beta (also in 1.6.6)

### OS version

macOS 13.5.2

### Additional system information

M1 MacBook Pro

### Printer

Ratrig Vcore3 500 (Klipper)

### How to reproduce

Use pattern or line version of the pressure advance calibration test and check first layer coordinates.

### Actual results

The first layer coordinates `first_layer_print_min` and `first_layer_print_max` are wrong. The print appear to occupy almost all the bed.

This is important because these variables are used by some START_PRINT macros in klipper. In my case these variables are used by an adaptive mesh ratio that only probes the actual print area to save time, but it could affect other things too.

This is the relevant part of the START GCODE:

`FAST_START EXTRUDER_TEMP=[first_layer_temperature] BED_TEMP=[first_layer_bed_temperature] X0={first_layer_print_min[0]} Y0={first_layer_print_min[1]} X1={first_layer_print_max[0]} Y1={first_layer_print_max[1]}`

It yields `FAST_START EXTRUDER_TEMP=245 BED_TEMP=105 X0=5 Y0=5 X1=495 Y1=495` in the actual GCODE.

X0=5 Y0=5 X1=495 Y1=495 is the complete bed minus 5mm from each border in my 500x500 printer.

### Expected results

The correct first layer coordinates should be passed. In my case, it should be about X0=190 Y0=210 X1=310 Y1=290 according to the real size of the print. The GCODE should look something like this:

`FAST_START EXTRUDER_TEMP=245 BED_TEMP=105 X0=190 Y0=210 X1=310 Y1=290`.

I started noticing this behavior after the introduction of the pattern PA test by @thewildmage in PR [#1547]. This happens both to the PA lines and pattern test, but nowhere else I'm aware off.

### Project file & Debug log uploads

No project file related.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Tolvelski, 2023-09-20:
I had the same issue last night. I was trying to use KAMP which depends on the model dimensions for things like the purge line and smart park.
When I check the gcode to the PA Test, both line and pattern, it was something like  POLYGON = [5, 5], [5, 295], [295, 5], [295, 295] which is my entire bed area. So when my machine tried to use the smart park or purge line from KAMP, which is supposed to be just outside the print area, I got an out of bounds error because the nozzle tried to move to [5, - 5].
In the end, to be able to run the PA Test I had to disable KAMP.

patrezp, 2023-09-21:
I can confirm this behaviour using KAMP with default polygon occupies whole bed and the print fails to start as it tries to go out of bounds. 

The fix for this is changing the POLYGON and min, max coordinates to something sane, I am using 120,120 240,240 on my voron 350
This way I get mesh and purge with KAMP enabled.

I can imagine some sort of rectangle with 0 infill being generated, that would go around the actual print, based on the amount of steps in the PA tests 

This way the min,max coordinates could be passed to the KAMP or other macros safely

closed 2023-12-29 by github-actions[bot]