OrcaSlicer/OrcaSlicer #2071: Windows Defender reports virus and blocks the EXE. (1.7.0 beta) (False alarm)

DokuKaefer, 2023-09-11:
### OrcaSlicer Version

1.7.0 beta

### OS version

Windows 11 22H2 Build 22623.1928)

### Printer

Bambu Lab X1C

### How to reproduce

Download the OrcaSlicer_Windows_Installer_1.7.0-beta.exe

### Actual results

Windows defender reports the exe as a virus and blocks it.
I think this is a false alarm.

[image]

### Expected results

No blocking of the exe file.

### Project file & Debug log uploads

no project

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2023-09-11:
I scanned it with VirusTotal, it's all green
here is the result: https://www.virustotal.com/gui/file/186fed7bca658d96b8091d37cf6d043a1e8251f8daf746289309330a01d07776

No idea why Windows defender suddenly has issue with the NSIS installer, it's the same installer script as previous release and is built in GitHub runner 😵‍💫

capsel22, 2023-09-11:
same here
[image]
[image]

Misunderstood-Wookiee, 2023-09-11:
> To me, this iimplies a strong suggestion of a false positive The Wacatac!ml/Wacapew!ml is a machine learning based detection, which is an AI based heuristics and behavioural analysis method in this case. They sound scary, but they're generic names. The detection on MaxSecure is known as a detection that can be false positive due to heuristic behavioural analysis and conservative behaviour The detection from Bkav Pro is an AI based detection, which is also heuristic behavioural analysis.
> 
> This is likely the case because the NSIS installer is so widely used, and could as such be used in a popular malware package. The AI's and machine learning algorithms simply learnt that the NSIS installer and some malware package(s) came bundled together.

Historically the program has been safe, I am inclined to agree with you, but also historically the program has not flagged Windows Defender nor Virus Total, I would err on the side of caution on this one until the dev can do a more meaningful analysis on their end.

SoftFever, 2023-09-11:
I will see if there is anything to upgrade for the NSIS.

Otherwise I clueless what else I can do.
The whole installer package is done by the same script as 1.6.6 on GitHub action runner, which is very unlikly contain virus...

DokuKaefer, 2023-09-11:
In this thread there are some options you can do.
But the good options are very expensive. 😬

https://stackoverflow.com/questions/48946680/how-to-avoid-the-windows-defender-smartscreen-prevented-an-unrecognized-app-fro

HumbleDeer, 2023-09-11:
I currently get no flags from Defender either; the definitions I have installed are from yesterday.

Misunderstood-Wookiee, 2023-09-11:
Not sure about others but my platform is Windows 11 insider prev build. Honestly, it is probably nothing.  
That said, nothing changed and here we are, though I have heard it has given false pos in the past.

l8nite, 2023-09-11:
Google chrome blocked this as well: 
[image]

l8nite, 2023-09-17:
For what it's worth the latest 1.7.0 release seems to have resolved this issue - this issue can probably be closed out.

scotty6435, 2023-11-01:
I've got the exact same issue with 1.8.0-beta on Windows 11:
[image]

jaegerschnitzel, 2023-11-12:
Same for me with all 1.8.x versions.

SimonThompson1965, 2023-11-23:
Me too, same issue using either the exe or the portable zip file
[image]

Boergen, 2024-02-19:
Issue is also still present in the current stable and 2.0.0 nightly builds.

closed 2024-05-27 by github-actions[bot]