OrcaSlicer/OrcaSlicer #2078: 1.7.0 Fedora Device page blacked out

thibaultvdb, 2023-09-11:
### OrcaSlicer Version

1.7.0

### OS version

Fedora 38

### Printer

RatRig VCore 400

### How to reproduce

After running the .appimage on fedora everything seems to work and the javascript bug seems resolved.

### Actual results

The device tab stays black and doesn't do anything.

### Expected results

The device tab should show the klipper interface.

### Project file & Debug log uploads

It's project independent.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

TexZeTech, 2023-09-12:
Raised in https://github.com/SoftFever/OrcaSlicer/issues/920

icheered, 2023-11-29:
@thibaultvdb Did you find a way to solve this? I'm running into the same thing.

closed 2024-03-06 by github-actions[bot]

TexZeTech, 2024-03-28:
Hey if either of you [@thibaultvdb](https://github.com/thibaultvdb) [@icheered](https://github.com/icheered) are running Nvidia drivers try https://github.com/SoftFever/OrcaSlicer/issues/920#issuecomment-2025895738 if not sorry to bother ya.