OrcaSlicer/OrcaSlicer #2079: "QoL thinner sliders" is not QoL for all

avxmw, 2023-09-11:
### OrcaSlicer Version

1.7.0 beta

### OS version

Linux AppImage

### Printer

ALL

### How to reproduce

1. install beta
2. slice

### Actual results

What is called a Quality of Life improvement is actually detrimental in my case as it is harder to see and to click/drag.

### Expected results

Actually be QoL improving by giving people an option to set as needed.

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

capsel22, 2023-09-11:
agreed. Let us set thin of fat slider in view menu

foreachthing, 2023-09-11:
Same here! On 4k monitors, it's just 6-7 pixel in height/width. Not QoL at all 😦

SoftFever, 2023-09-11:
Dman, I forgot the DPI scale is a thing.

Are you all running on 4K motinor?

SoftFever, 2023-10-01:
Fixed.

closed 2023-10-01 by SoftFever