OrcaSlicer/OrcaSlicer #2081: Layer proceeding a bridge is not oriented optimally

samwiseg0, 2023-09-11:
### OrcaSlicer Version

1.7.0-beta

### OS version

MacOS 13.5.1 (22G90)

### Additional system information

N/A

### Printer

Voron 2.4

### How to reproduce

1. Add primitive (Voron Cube)
2. Slice with walls greater than 1
3. Observe that the preceding layer after a bridge follows the same orientation as the bridge itself. 

### Actual results

Bridge layer
[image]

 The preceding layer after a bridge follows the same orientation as the bridge itself. 
[image]

If you reduce walls to 1 then you get the desired result. 
[image]

I have also tried to use classic vs Arcane and it has the same result.

What is produced on the print itself. Since the layer above the bridge is in the same orientation and it is not considered a bridge. The extrusion speed, cooling etc is not applied. Which can cause undesired results if your bridge density is not very high (>80%). 
[image]

### Expected results

The layer above the bridge uses an offset angle from the bridge or uses the infill angle.

As stated above using only a single wall gives you the desired results.

This is an example from SuperSlicer.
[image]

What is produced printed with SuperSlicer.
[image]

I apologize if I missed a setting somewhere that would produce the desired result. I have tried a few things and could not get bridges to print as well as in SuperSlicer. 

### Project file & Debug log uploads

[Voron_Design_Cube.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12580552/Voron_Design_Cube.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2023-09-13:
Untick detect narrow internal solid infill option and it should print properly.

igiannakas, 2023-09-13:
[image]

samwiseg0, 2023-09-14:
@igiannakas Thank you for pointing that out. I have no idea how I missed that checkbox! 

Thanks again!

samwiseg0, 2023-09-14:
p.s I believe unchecked should be the default? For the most part I cannot see a reason that it should be on by default?

closed 2023-09-14 by samwiseg0

igiannakas, 2023-09-14:
> p.s I believe unchecked should be the default? For the most part I cannot see a reason that it should be on by default?

Speed… concentric infill is faster to print in small areas. But it will result in a slightly worse bridge. I’ve disabled it on my profiles btw

igiannakas, 2023-09-14:
I've published a PR to Softfever to make the logic behind setting concentric infill less sensitive so more bridges are bridged properly vs. the concentric infill being used. Hopefully this gets published so we can use both a correct bridge behaviour and the faster concentric infill in areas where small infill is present.