OrcaSlicer/OrcaSlicer #2083: Ankermake printers do not appear

Matrekus, 2023-09-12:
### OrcaSlicer Version

1.7.0

### OS version

WIN 11

### Printer

Ankermake M5C&M5

### How to reproduce

Select one of the Ankermake printers when start the setup assistant or in Orca afterwards. 

### Actual results

The printer does not appear in the list and cant be selected.

### Expected results

Printer should be selectable in the printer dropdown.

### Project file & Debug log uploads

If you copy the profiles from bambu studio it works like it should.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

UncleJessy, 2023-10-10:
Confirming this is also an issue with Mac OS Orca 1.7.0

[image]

[image]

eliezedeck, 2023-11-08:
I've had this problem many times, even with the latest 1.8.0-beta1.

For other printers, it seems to appear immediately. For AnkerMake printers, you'll need to select the printer, then close Orca. Then open again, now it will appear.

closed 2024-02-20 by github-actions[bot]