OrcaSlicer/OrcaSlicer #2085: Allow Extruder Clearance to have a different value in 4 directions (+X, -X, +Y, -Y), especially for BBL X1C with large LIDAR module on one side.

qeggs, 2023-09-12:
**Is your feature request related to a problem? Please describe.**
I would like to place objects closer together for printing sequentially "by object". Unfortunately, a large distance of 68mm is required to separate components to prevent the LIDAR module on the BBL X1C from crashing into adjacent objects which have already been printed.

**Which printers will be beneficial to this feature**
BBL X1C will be benefitted because it has a large LIDAR module which extends out 68mm on one side of the toolhead. The other sides of the toolhead extend less than 50mm.

**Describe the solution you'd like**
Would like to be able to set the Extruder Clearance "Radius" to a different value to the +X, -X, +Y, and -Y directions. 

**Describe alternatives you've considered**
I can currently set the Radius to any number. If I place parts in positions in close-packed positions, then I can trick the software into working by setting the Clearance Radius to a very small number. This removes any protection that the software gives. I must be very careful to considering the object printing order to ensure there will be no crashes. 

**Additional context**
Extruder Clearance is setting on the configuration page for the Printer and Nozzle (screenshot)
[image]
[Extruder Clearance Demo2.zip](https://github.com/SoftFever/OrcaSlicer/files/12582255/Extruder.Clearance.Demo2.zip)

The attached file shows that the Slicer software gives a warning and will not allow the plate to be sliced. However, if the extruder clearance radius is reduced, then this will slice. Physically, this plate would print OK, without any toolhead crashing, because the left object would print first and then the right object.

TexZeTech, 2023-09-12:
I would rather be able to make a 3D model of the print head so it can base the clearances on that. idk.

foreachthing, 2023-12-12:
A radius is not very useful since my nozzle is not symmetrical centered to the print head, so it should be XY in both directions, that will be really helpful.

fbson, 2024-04-08:
It really would be nice if anyone does have a way of doing this - it would greatly improve efficiency and throughput for a lot of applications I reckon. Thanks if you can help with it!

hxpphx, 2024-04-12:
I'd like to be able to set the Extruder Clearance with a rectangular shape instead than circular. I my opinion it will better if I could specify the distance from nozzle in front, rear, left and right directions.

fbson, 2024-07-01:
It's a little quiet about this one, and remains something that I for one would love to have as a feature of Orca. I'm assuming it's a 'no' then? Thanks in hope...!

Hammerfest, 2024-08-31:
> I'd like to be able to set the Extruder Clearance with a rectangular shape instead than circular. I my opinion it will better if I could specify the distance from nozzle in front, rear, left and right directions.

true, and with 3d printers now having more elaborate toolheads these days, needed with pretty much all of the printers I have, sadly, they are frying other fish ATM looking at the commits, push's etc

malventano, 2024-10-09:
Just had a part punted off the bed by the lidar, even though I was using the default radius and all objects appeared clear of each other in the slicer, so here's another vote for x/y dimensions for extruder clearance :).

closed 2025-01-16 by github-actions[bot]

malventano, 2025-01-16:
> [github-actions](https://github.com/apps/github-actions)
closed this as [not planned](https://github.com/SoftFever/OrcaSlicer/issues?q=is%3Aissue%20state%3Aclosed%20archived%3Afalse%20reason%3Anot-planned)[16 hours ago](https://github.com/SoftFever/OrcaSlicer/issues/2085#event-15949497863)

Why?

foreachthing, 2025-01-18:
Please reconsider. The nozzles of my printers are not in the center of the print head.

ghyker, 2025-02-09:
This issue must be solved. Not having a true shape for the extruder head essentially reduces the size of my machines build volume. The -Y- Volume is reduced by 37mm (68mm - 31mm) on your -Y- negative side per -Y- row count and additionally the -Y- Plus side is reduced by 12mm (68mm - 56mm) ... per row. Total is 49mm per row. Not good!!
Now for the -X- we are shorted by 67mm (68mm*2 - 59mm) per column of parts.
Please put in a couple hours and resolve this.

fbson, 2025-02-10:
Agreed, please do not close or not plan this - it would be a great improvement to the functionality of the software and would represent a big advantage when compared to other software too. Go on, strive to improve it...!

stickmansam4, 2025-03-04:
this is a basic option already present in Cura. so far i have come across no print head that is symmetrical enough to dictate this restriction. PLEASE give us seperate XY "rectangle" dimensions.

Tida87, 2026-02-20:
This would be greatly needed. Practically all bambu printers have unique dimensions in several dimensions.

[image]