OrcaSlicer/OrcaSlicer #2086: Linux -- Process - Objects - Text black.

TexZeTech, 2023-09-12:
### OrcaSlicer Version

1.6.6 and 1.7.0 beta

### OS version

6.1.51-1-MANJARO (64-bit)

### Additional system information

Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.7
KDE Frameworks Version: 5.109.0
Qt Version: 5.15.10
Kernel Version: 6.1.51-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-3770K CPU @ 3.50GHz
Memory: 15.6 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080/PCIe/SSE2

### Printer

Custom profile. #Customized Robo3dR1+

### How to reproduce

1. Make sure the system is in dark mode
2. have `"dark_color_mode": "1",` in the OrcaSlicer.conf  (shouldn't matter)
3. Select printer profile
4. Select filaments
5. Have Advanced enabled (shouldn't matter)
6. Select Objects button

### Actual results

[image]

[image]

### Expected results

Readable text.

### Project file & Debug log uploads

[OrcaCube.zip](https://github.com/SoftFever/OrcaSlicer/files/12582858/OrcaCube.zip)

[debug_Mon_Sep_11_23_55_52_27964.log.0.log](https://github.com/SoftFever/OrcaSlicer/files/12582864/debug_Mon_Sep_11_23_55_52_27964.log.0.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

TexZeTech, 2023-09-12:
Also present in the Bed shape dialog

[image]

Extruder tab -> Extruder offset

[image]

TexZeTech, 2023-09-12:
Also in the network window
[image]

mdziekon, 2023-10-04:
I've experienced the same problem, and since so far there was no response from the maintainers, also decided to look myself into the issue.

So far I've been able to determine the following things:
- Changing back & forth between system's dark mode preference seems to "fix" some of the labels. When previously they would be black on dark background, after changing from dark, to light, to dark again, the label is now white-ish, as one would expect.
- Forcefully setting foreground color in some places (eg. here https://github.com/SoftFever/OrcaSlicer/blob/8e700818796c12b7c66fe7d4443956d70a68410e/src/slic3r/GUI/GUI_ObjectList.cpp#L93) seems to fix the problem
  - However, I'm not sure if this is the right approach. Looking at PrusaSlicer's code in corresponding places does not indicate any need for such things.

@SoftFever do you have any thoughts about the second point from my findings? If you think this is an acceptable approach, I can create a PR with such changes, for any place I can find where it fixes the dark mode.

SoftFever, 2023-10-04:
> I've experienced the same problem, and since so far there was no response from the maintainers, also decided to look myself into the issue.
> 
> So far I've been able to determine the following things:
> 
> * Changing back & forth between system's dark mode preference seems to "fix" some of the labels. When previously they would be black on dark background, after changing from dark, to light, to dark again, the label is now white-ish, as one would expect.
> * Forcefully setting foreground color in some places (eg. here https://github.com/SoftFever/OrcaSlicer/blob/8e700818796c12b7c66fe7d4443956d70a68410e/src/slic3r/GUI/GUI_ObjectList.cpp#L93
>   ) seems to fix the problem
>   
>   * However, I'm not sure if this is the right approach. Looking at PrusaSlicer's code in corresponding places does not indicate any need for such things.
>   * Also, to be clear, the mentioned place is just a marker. It still needs something like `this->SetForegroundColor()` to actually set the color of the font.
> 
> @SoftFever do you have any thoughts about the second point from my findings? If you think this is an acceptable approach, I can create a PR with such changes, for any place I can find where it fixes the dark mode.

Hi @mdziekon 
Thanks for investigating the issue.
I don't have clues either unfortunately.
I'm open to using workaround solutions. 
Given that this is a Linux-specific issue, I'd suggest wrapping the codes with #ifdef __linux__  to minimise the risk

mdziekon, 2023-10-07:
@TexZeTech @guestisp @GAZ082
Since all of you reported similar issues, and you all have different distros in use (I'm personally on ancient Ubuntu 20.04 with Gnome 3.36.8, X11, yaru-dark theme), would anyone of you care to compile my PR & test my partial fix? It does fix the issue for me, but it would probably be nice to confirm that it does so for other people as well.

SoftFever, 2023-10-07:
I have kicked off the CICD build if that helps.

guestisp, 2023-10-07:
Trying the automated build on Ubuntu. there is an issue with the initial checkbox but everything else looks fine to me, i'm able to see the texts. (not sure at 100% because i've never used this slicer before, due to missing texts and labels)

[image]

guestisp, 2023-10-07:
Uhm, some issues are still present here
[image]

mdziekon, 2023-10-07:
@guestisp as per pull request's title, my recent change fixes **only** the object's list, and nothing else. I couldn't find a quick solution for other places, so I'm fixing (or at least, trying) one place at a time.

guestisp, 2023-10-07:
Sorry i didn't read the title carefully.

Object list is working properly, in 1.7 was black on dark grey, now is white on dark grey as expected

GAZ082, 2023-10-07:
@mdziekon where is the build with this patch, couldnt find it.

Thanks!

mdziekon, 2023-10-07:
@GAZ082  I think this should be it: https://github.com/SoftFever/OrcaSlicer/actions/runs/6438345400?pr=2346

guestisp, 2023-10-07:
yes, that

TexZeTech, 2023-10-07:
> @GAZ082 I think this should be it: https://github.com/SoftFever/OrcaSlicer/actions/runs/6438345400?pr=2346

[image]

Just running that appimage as is seems to fix the one thing, it doesn't seem to fix the other dialogs.

TexZeTech, 2023-10-07:
I should probably also rename this to "Linux -- darkmode ux issues" instead IDK.

GAZ082, 2023-10-07:
Yes, this fixed the objects dialog, partially.

[image]

kesor, 2023-10-22:
This issue makes OrcaSlicer pretty unusable in Linux. It is still possible to override the dark mode in GTK to be light-mode just for OrcaSlicer by setting the `GTK_THEME=:light` environment variable.

For example, using a script like this -

```shell
#!/bin/sh

export GTK_THEME=:light
exec OrcaSlicer
```

guestisp, 2023-10-28:
As this is a semi-blocking bug on Linux, why not forcing the light theme directly in the app image ? Better a white working theme than dark but unusable

TexZeTech, 2023-10-31:
As of [OrcaSlicer V1.8.0 Beta Release ](https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.8.0-beta) the object text is no longer black, but it still has issues with the text on the tabs. 

[image]

In FillamentSettings -> Multimaterial -> Ramming customization
we have text that is black on black (or gray on gray?)
[image]

In the Compare presets dialog 

There is a checkbox.... I have no idea what it says.

[image]

Filament Color dropper is black on grey (this is not a big deal)

[image]

Flushing Volumes has grey on grey as well.

[image]

--Printer Settings--

Basic information -> Best object position 

[image]

Basic information -> Printable Area -> Set dialog

[image]

Extruder -> Extruder Offset

[image]

--Network--

Text stating what the input fields are for is unreadable, input for most is usable with the exception of the profile at the top.
[image]

Searching for the network is also in need of some TLC.

[image]

--Save/Discard changes--

[image]

This technically fixes the "Object - Text black" so if a dev wants me to close this for that bug fix and open a new one for the other things that need to be fixed I will.

Thank you again for the work.

TexZeTech, 2023-10-31:
> As this is a semi-blocking bug on Linux, why not forcing the light theme directly in the app image ? Better a white working theme than dark but unusable

You're correct but I really dislike light themes. If it was truly unusable I would go to the effort of swapping it to light mode for just this application.

guestisp, 2023-10-31:
i hate light themes too but Orca is unusable on Ubuntu
so, why not forcing a light theme (until a fix) or why not adding a setting to change the theme from the app?

TexZeTech, 2023-10-31:
Confirmed that it's broken the exact same way on Ubuntu 22.04.03 as my system with dark mode.

TexZeTech, 2023-12-01:
This all seems to be fixed on the current release on my laptop so I think something is fubar with my tower install.

closed 2023-12-01 by TexZeTech