OrcaSlicer/OrcaSlicer #2095: HUGE slicing speed difference between X1C and P1S - plus Variable Layer Height support distance error

titaniumschmapple, 2023-09-12:
### Discussed in https://github.com/SoftFever/OrcaSlicer/discussions/2094

<div type='discussions-op-text'>

<sup>Originally posted by **titaniumschmapple** September 12, 2023</sup>
I've got the same file and same settings and I choose the X1C and slice - 1h 20m

Change to the P1S and slice - 2h 35m

I'm changing nothing but the printer and it's nearly doubling the print time but on Bambu Studio it's the same print time, is this an OrcaSlicer error?

Also by the way, on a separate issue, the Top Z Distance is incorrect when using Variable Layer Height and Tree (Auto) and Organic Supports - it makes the Z Support Distance huge, over 1mm (instead of 0.08mm), got to be an error too right?</div>

igiannakas, 2023-09-12:
Organic supports don’t work with variable layer height. It’s a limitation in the prusa implementation that has been carried over.

shaneshort, 2023-09-13:
@titaniumschmapple can you provide a screenshot showing where the time difference is coming from and maybe a link to the model you're using to help the developers replicate your issue?

closed 2023-12-21 by github-actions[bot]