OrcaSlicer/OrcaSlicer #2099: Add a Z offset function

jiiimi, 2023-09-13:
Is it possible please to add a Z offset function like in Prusaslicer (Printer settings/General/Z offset)?

shaneshort, 2023-09-13:
Is this not something you could implement with custom start gcode?

Can you explain your use case a little better?

igiannakas, 2023-09-13:
You can do that directly in the print start gcode or in the filament gcode if you need to make it filament specific.

gak45, 2023-09-14:
Yeah but that is a pain in the ass if you work with multiple types of filament on multiple types of beds.

jeremytodd1, 2023-09-14:
I've been doing this in the filament's custom gcode section but I fully agree that it'd be nice if it was an actual setting in the filament profiles.

Why should we settle for forcing people to do it with gcode? Not everyone knows gcode. If Orca is trying to be a very useful friendly slicer then this feature is a no brainer in my mind. It's a simple thing but is a nice quality of life feature.

I don't think "just do it in gcode" is a legitimate answer to this feature request.

kylek29, 2023-09-14:
> Yeah but that is a pain in the ass if you work with multiple types of filament (try crushing TPU!) on multiple types of beds (z offset should vary from smooth vs textured vs engineering).

Maybe it should be added to some form of Bed config overrides as well? I noticed that my X1C tends to dig into my Whambam plate with certain plates selected. I suspect it's because they use the LIDAR to do some of the pre-print ABL and it's not properly detecting the clear PEX sheet.

gak45, 2023-09-14:
If you are using the textured plate QR code its line 274 (":;===== for Textured PEI Plate , lower the nozzle as the nozzle was touching topmost of the texture when homing ==") to 278("{endif}") that is causing it to dig (and is the 'just do it in g code' code that certain snobs are talking about). Dial back the extra z adjustment via line 277's -0.0x amount specified. OR. remove that block of code all together.

shaneshort, 2023-09-15:
> I've been doing this in the filament's custom gcode section but I fully agree that it'd be nice if it was an actual setting in the filament profiles.
>
> Why should we settle for forcing people to do [...]

Sorry, I guess I wasn't super clear in my reply.

Yeah, seems like a really good idea and something that couples with a feature request I've been meaning to make which would be to unlock the bed profiles for printers other than bambu.

What I was trying to say albeit poorly, was: "in the meantime, you can do this in gcode, that's how I do it currently"

R4toLoko, 2023-09-15:
I think you guys are after this feature request I added today.
I solved this on my Voron with custom g-code atm. It works amazingly well with my 5 sheets but there is room to make this more user friendly.

https://github.com/SoftFever/OrcaSlicer/issues/2120

ju5th3r3, 2023-09-24:
Please Add z offset setting to be user friendly. I actually lower my nozzle to the bed then, adjust the z offset in profiles in cura to compensate for different filaments. this works great just wish I had a user friendly setting in orca as I really want to make it a go to slicer but needs this function so my other machines as easy to adjust. So Please Please add this feature

Ursus63, 2023-10-31:
I need z-offcet too. It's very usefull option for me in Prusa and Superslicer. I dont understant why there is no this option in Orca. I can understand why Bambu studio don't have(all printers with ABL and other bed automation). But Orca is a universal slicer of any printers, and still not all printers have ABL or smthg like that. And z-offcet is very usesfull for such printers.
And it whould be awsome if z-offcet become an option for filament but not a printer setting as it in PS or SS.

igiannakas, 2023-10-31:
> I need z-offcet too. It's very usefull option for me in Prusa and Superslicer. I dont understant why there is no this option in Orca. I can understand why Bambu studio don't have(all printers with [...]

https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.8.0-beta

Its implemented in v1.8.

jeremytodd1, 2023-10-31:
> Its implemented in v1.8.

Not quite. Sure, a GUI Z offset option has been added to the printer profiles, but it is still missing from the filament profiles. The only way to set z offset per filament is still only through gcode.

Being able to set the offset per filament is pretty important, as different filament types use different z offsets.

R4toLoko, 2023-10-31:
> Being able to set the offset per filament is pretty important, as different filament types use different z offsets.

Wrong!

A printer has a single off-set per sheet/bed, not filament.
Different filament types requires different flow calibration, this is the first time within years I hear somebody saying offset is done my filament and not by sheet/bed

jeremytodd1, 2023-10-31:
Ok, bud. Not here to argue. It's pretty commonly known, even after proper flow adjustment, that filament like PETG requires a different z offset than ABS, for example.

Either way, it'd be a good idea to add a GUI z offset item in the filament profiles as well. It's a simple change (it's also one that is easily ignored if they don't want to use it) and would allow people more control over their specific setups in an easier way.

Ursus63, 2023-10-31:
> > Being able to set the offset per filament is pretty important, as different filament types use different z offsets.
> 
> Wrong!
> 
> A printer has a single off-set per sheet/bed, not filament. Different filament types requires different flow calibration, this is the first time within years I hear somebody saying offset is done by filament and not by sheet/bed

Today i wanted to print ABS on my printer that have perfect z height for PLA and PETG(didn't use  ABS on this printer for some years) and nozzle scratches my build plate. How flow calibration should help me in that situation?

R4toLoko, 2023-10-31:
> and nozzle scratches my build plate

A filament will NEVER cause that. The first layer will over or under extrude until you calibrate the filament profile but never do what you said.
The only way a nozzle scratches a build plate is if the bed profile offset is incorrect or like in my case with 5 sheets = 5 offset, picked the wrong profile. My Textured offset is -0.245 while my Smooth is 0.04, pick the wrong profile and rip build plate.

I don't think we are talking about the same thing here, period.

Ursus63, 2023-10-31:
> A filament will NEVER cause that. >

Filament no. And what about heatbed temperature?
You don't know that heatbed can have a different shape on different temperatures? What happened in that case if you don't have ABL?

R4toLoko, 2023-10-31:
> Filament no. And what about heatbed temperature?
> You don't know that heatbed can have a different shape on different temperatures? What happened in that case if you don't have ABL?

I only use steel PEI sheets like many people do, my PEI Textured offset is one for both for printing PLA and ASA. My ASA is calibrated based on that offset, not the other way around.
If your bed heat expansion is being that great enough to make the nozzle to hit the bed, something isn't right.
Anyway, this is not the topic of this ticket, two different problems.

skl111, 2023-11-13:
> Wrong!
>
> A printer has a single off-set per sheet/bed, not filament. Different filament types requires different flow calibration, this is the first time within years I hear somebody saying offset is [...]

Just because you haven' heard about it, doesn't mean it hasn't been done. Cura has this as an option for years.

https://github.com/fieldOfView/Cura-ZOffsetPlugin

rafa-paint, 2023-12-11:
Hola, estoy usando orca slicer y me gustaría saber qué gcodes usáis en el apartado de filamento para ajustar un zoofset para ese filamento y una vez terminada la impresión vuelva al x offset de la impresora. Gracias

JeffreyOI, 2024-01-09:
> Wrong!
>
> A printer has a single off-set per sheet/bed, not filament. Different filament types requires different flow calibration, this is the first time within years I hear somebody saying offset is [...]

Sheet type is certainly an important factor in good z-offset calibration, but so is filament type. Filament-specific z offset is definitely a thing. See WhamBam's explanation for PETG, 1:15: https://youtu.be/8839jEA-KQc.

zionlion77, 2024-02-01:
@HakunMatat4 I don't agree with you on your argument that the printer and bed combination determines the z-offset and not the filament. You can find plenty references, how different types of filament need different amount of squish. The video that @JeffreyOI shared explains it nicely for PETG.

My personal experience with my Ender 5 Pro and my Bambu Lab X1C using textured PEI build plates on both machines confirms this. A typical z-offset of around 0.1 mm (0.4 mm nozzle) works great for me to get perfect first layers with PLA. Yet, if I used that offset with PETG without using glue stick or something similar, I can trash my build plate after the print, because the PETG is squished so hard into the PEI that both materials bond.

Therefore, I would also love to see a z-offset override feature as part of the filament settings. Until we will get something like that in Orcaslicer, I use the following code in the "Machine start G-code" section, right after the `;===== for Textured PEI Plate (...)` block

```
;===== lower bed for PETG
{if filament_type[initial_no_support_extruder]=="PETG"}
G29.1 Z{nozzle_diameter[initial_no_support_extruder]*0.25} ; increase bed nozzle distance by a quarter of the nozzle diameter
{endif}
```

You can play around and optimise the fraction of the nozzle diameter to add in z according to your filament and overall print settings.

ju5th3r3, 2024-02-02:
Where do I enter the code?
This is an older marlin firmware machine. Qidi Xmax (the original one)
I selected different beds and this is what I have in filament section
[image]

zionlion77, 2024-02-03:
as I said, I didn't place the code within the "Filament start G-code", but inside the "Machine start G-code" section under printer settings, right after the `;===== for Textured PEI Plate (...)` block. The filament start g-code gets executed too early, that's why I did it as described.

ju5th3r3, 2024-02-04:
@zionlion77 that section is blank for me so I didn't see ;===== for Textured PEI Plate (...) in either spot.
[image]

ju5th3r3, 2024-02-04:
I do see this under basic information
[image]

closed 2024-05-12 by github-actions[bot]