OrcaSlicer/OrcaSlicer #2101: Printer / Filament / Process dropdown menus do not display

back2ll, 2023-09-13:
### OrcaSlicer Version

1.7.0-Beta

### OS version

Debian 12

### Printer

Creality K1

### How to reproduce

Installed 1.7.0-Beta AppImage on Debian 12

### Actual results

The menus to select a printer or filament or process do not display. Only a blank list or none at all.  (Similar to this issue: https://github.com/SoftFever/OrcaSlicer/issues/1875). It also happened with the previous version.

[image]

### Expected results

Display menus

(Works fine on Ubuntu 22.04)

### Project file & Debug log uploads

[Uploading log.zip…]()

### Checklist of files to include

- [ ] Log file
- [ ] Project file

theotherjimmy, 2023-11-02:
I have this issue too, I'm on Nixos-Unstable.

jmickelin, 2023-11-15:
Same here. Tried both the 1.7.0 and the 1.8.0-rc2 AppImages. It should not matter, but for reference: my window manager is i3, running on Fedora 38.

jmickelin, 2023-11-15:
Going back trough the releases, the bug is present as far back as OrcaSlicer 1.5.0, but _not_ present in BambuStudio 1.4.5. So the bug was introduced between those two releases.

314159-r, 2023-12-28:
I have the same issue - **All** dropdown controlls are broken, all over the entire application.
Appimage ver. 1.8.1 on Fedora 38 and Debian 11, both with xfce as window manager.
(On Fedora I needed to install webkit2gtk4.0 to get it to start)

Totally bricks the app...

314159-r, 2024-01-01:
It works fine on my (very) old laptop, running X-ubuntu 22.04, also with xfce desktop.

jmickelin, 2024-04-06:
Still present in 2.0.0.

jmickelin, 2024-04-06:
As an added sanity check, it is also present in both the AppImage releases named "Fedora" and "Ubuntu" of BambuStudio [v01.09.00.60](https://github.com/bambulab/BambuStudio/releases/tag/v01.09.00.60).

As previously mentioned, it was not present in 1.4.5. I would offer to Git bisect it for you to find the offending commit that introduced the regression, but I have not had much luck compiling the program myself.

Just from a cursory reading without testing it out, there seems like there were two potentially suspect commits that muck about with the popup code in between 1.4.5 and the first broken release in the 1.5.0 release cycle:

https://github.com/SoftFever/OrcaSlicer/commit/a03dc779b686a2b4761b0e11db13c650b38ac748
https://github.com/SoftFever/OrcaSlicer/commit/60dbf71cd6204ffc3b24e4369a66bc3f0b533db2

closed 2024-07-14 by github-actions[bot]

jmickelin, 2025-01-14:
Fixed upstream in https://github.com/bambulab/BambuStudio/pull/5704

back2ll, 2025-01-14:
Thank you for doing this