OrcaSlicer/OrcaSlicer #2103: Support Upload and Print feature on Creality K1

victornpb, 2023-09-13:
I really want this feature on Orca slicer
[image]

I tried just setting up as a Octoprint/Klipper and giving it an IP, the device tab works but It doesn't understand the API.

What does the API normally looks like?

This is what uploading a gcode file on the web interface looks like:

It is just a POST on `/upload/filename.gcode` endpoint
```sh
curl 'http://192.168.0.104/upload/K1%20Exaust%20Fan%20Muffler%20(2)_PLA_1h45m.gcode' \
  -H 'Accept: application/json, text/plain, */*' \
  -H 'Accept-Language: en-US,en;q=0.9,pt;q=0.8' \
  -H 'Connection: keep-alive' \
  -H 'Content-Type: multipart/form-data; boundary=----WebKitFormBoundary7cSZUxfFJUfMXs3r' \
  -H 'Origin: http://192.168.0.104' \
  -H 'Referer: http://192.168.0.104/' \
  -H 'User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/115.0.0.0 Safari/537.36' \
  --data-raw $'------WebKitFormBoundary7cSZUxfFJUfMXs3r\r\nContent-Disposition: form-data; name="file"; filename="K1 Exaust Fan Muffler (2)_PLA_1h45m.gcode"\r\nContent-Type: application/octet-stream\r\n\r\n\r\n------WebKitFormBoundary7cSZUxfFJUfMXs3r--\r\n' \
  --compressed \
  --insecure
```

Starting a print is done via websocket, this is where the ws is

```sh
curl 'ws://192.168.0.104:9999/' \
  -H 'Pragma: no-cache' \
  -H 'Origin: http://192.168.0.104' \
  -H 'Accept-Language: en-US,en;q=0.9,pt;q=0.8' \
  -H 'Sec-WebSocket-Key: hujCwV9dGN8BJF+uYwObAg==' \
  -H 'User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/115.0.0.0 Safari/537.36' \
  -H 'Upgrade: websocket' \
  -H 'Cache-Control: no-cache' \
  -H 'Connection: Upgrade' \
  -H 'Sec-WebSocket-Version: 13' \
  -H 'Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits' \
  --compressed
```

And this is the actual message:
```json
  {"method":"set","params":{"opGcodeFile":"printprt:/usr/data/printer_data/gcodes/K1 Exaust Fan Muffler (2)_PLA_1h45m.gcode"}}
```

[image]

Does this looks like something that could be easily implemented? If you need more details on anything including running tests I'll be happy to help

brentrichardjr, 2023-09-17:
From my experience setting up and rooting my friend's K1 Max, I suspect your using stock firmware, in which case that would be a creality limitation due to creality's custom klipper and associated software. The rooted firmware works, although it's best to only slice and upload in orcaslicer and then start it on the printer or in creality print.

victornpb, 2023-09-18:
Yes, my original post is about supporting unmodified creality firmware, this would benefit a lot of people currently using creality print slicer which is mostly non developers.

closed 2023-12-26 by github-actions[bot]