OrcaSlicer/OrcaSlicer #2104: X1C Camera Error - BambuSource.dll

xenon2008, 2023-09-13:
### OrcaSlicer Version

1.7.0 beta & 1.6.6

### OS version

Windows 10 Pro x64

### Printer

Bambulab X1C Combo

### How to reproduce

I downloaded OrcaSlicer and added my brandnew X1C Combo Printer.
I installed the Bambu Network Plugin und logged in into my BambuLab Account
Then i switch to Device Tab and want to try the Camera Feed.

I got an Messaged that the BambzSource has not correctly registered... => i Clicked yes and got an Error again.
(The folder path is correct and the files are there too)
also tried to Download Bambustudio & copied the files from them to my OrcaSlicer Plugins Folder - same Result
Also with the latest 1.6.6 stable version

Here some Screenshots:
[image]
[image]

Could somebody help me?
Its my first Bambulab Printer and did not one testprint with it  =) 

Kindly Regards
Xenon

### Actual results

See the two Error Screenshots above

### Expected results

I thought i could see the Camera Stream/Feed ;)

### Project file & Debug log uploads

Files in Plugin Folder
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2023-09-13:
Download and install the official slicer - maybe there is a conflict somewhere. Then retry installing orca.

xenon2008, 2023-09-15:
yes that looks good. 
I dont unterstand why., because i already tried to copy the original Files from BambuStudio to OrcaSlicer....
Only difference now is, that the registered DLL in Windows links to "C:\Users\USERNAME\Appdata\Local\BambuStudio" and not to the FilePath of OrcaSlicer, but it works. => So why not :D

closed 2023-12-23 by github-actions[bot]