OrcaSlicer/OrcaSlicer #2107: Don't Support Bridges not available for tree supports

sdooweloc, 2023-09-14:
### OrcaSlicer Version

1.6.6 and 1.7.0-beta

### OS version

Arch Linux

### Additional system information

Kernel: 6.4.12-zen1-1-zen 
CPU: AMD Ryzen 7 7700X (16) @ 4.500GHz 
GPU: AMD ATI Radeon RX 6700/6700 XT/6750 XT / 6800M/6850M XT 

### Printer

Voron 2.4

### How to reproduce

Navigate to Support with Advanced enabled
change support type to tree 
and subtype to any non-organic option
observe lack of "don't support bridges" option. 

### Actual results

When you select tree supports with any "tree" style the option for "Don't support bridges" is no longer available for toggling and thus unnecessary supports get generated

### Expected results

I would expect the "don't support bridges option to be available for all support types and styles to prevent needless support. 

### Project file & Debug log uploads

I am not providing a zip because it happens on all profiles and it would be faster to just toggle the option for the tester than to download, extract, and import a zip. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2023-09-15:
"Don't Support Bridges" doesn't work for slim trees from the very beginning. Use `max_bridge_length` instead.

sdooweloc, 2023-09-15:
ah okay thank you for that information. I will close this issue accordingly.

closed 2023-09-15 by sdooweloc

mrelwood, 2023-09-19:
"Don't support bridges" is found with the search anyways, so it's confusing that it doesn't point to an option in the app.

Noisyfox, 2023-09-19:
> "Don't support bridges" is found with the search anyways, so it's confusing that it doesn't point to an option in the app.

That's true for all hidden options. I guess it can be improved by adding a label prefix like "Disabled" in search results for any options that are not appliable.