OrcaSlicer/OrcaSlicer #2112: Implement CTB Files

NickSica, 2023-09-14:
**Is your feature request related to a problem? Please describe.**
I developed a pull request for PrusaSlicer to implement CTB files for Elegoo based printers. Since this is a fork of a fork of PrusaSlicer that is more receptive to PRs I was recommended to bring my work here. I was wondering how much of the resin side of the slicer is still available in OrcaSlicer and how big of an ordeal it would be to try and bring this feature over(if you know off the top of your head). The PR is [here](https://github.com/prusa3d/PrusaSlicer/pull/10531).

**Which printers will be beneficial to this feature**
Elegoo Printers

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
Close sourced slicers.

kozross, 2023-11-05:
From a casual glance, I don't think _any_ resin printing stuff made it over. While I don't resin print, one big blocker for me was lack of open-source slicer support, so I would be _very_ interested in this landing if possible.

NickSica, 2023-11-06:
I have paper deadlines so I don't have much free time to work on this atm. When I have time, I'd like to finish the updates I made to turboresin then move onto working on the best solution for incorporating my changes either into this or a PrusaSlicer fork. If anyone would like to try their hand at this in the meantime, I am open to answering any questions (if I know the answer).

kozross, 2023-11-06:
@NickSica - mostly for my own edification: your changes support a specific kind of CTB format, which is mainly what Elegoo printers run. What other printers from other manufacturers could this potentially open up support for?

NickSica, 2023-11-06:
> @NickSica - mostly for my own edification: your changes support a specific kind of CTB format, which is mainly what Elegoo printers run. What other printers from other manufacturers could this potentially open up support for?

If using all the changes from PrusaSlicer, I think also any printers supporting pwmx. I tried to make my version as generic, but it is also very much on the brute force side of things. I'm wondering if there is a better way to do things, but I'm not sure with all the bs resin printer manufacturers pull.

kozross, 2023-11-07:
> > @NickSica - mostly for my own edification: your changes support a specific kind of CTB format, which is mainly what Elegoo printers run. What other printers from other manufacturers could this potentially open up support for?
> 
> If using all the changes from PrusaSlicer, I think also any printers supporting pwmx. I tried to make my version as generic, but it is also very much on the brute force side of things. I'm wondering if there is a better way to do things, but I'm not sure with all the bs resin printer manufacturers pull. I did want to implement the GOO file format since that seems like an open source file format that elegoo might stick with from here on out but only time will tell.

Could you tell me which printers are known to support pwmx? I'm really unfamiliar with this side of things.

NickSica, 2023-11-07:
> Could you tell me which printers are known to support pwmx? I'm really unfamiliar with this side of things.

I'm also not 100% sure, I think it's some of the anycubic printers? I think the Mono X and newer.

closed 2024-02-20 by github-actions[bot]