OrcaSlicer/OrcaSlicer #2114: Disable Set remaining print time

CCS86, 2023-09-15:
Could we please have an option to disable the gcode output for M73: Set remaining print time?

foreachthing, 2023-09-15:
You could write a small post-processing script which would remove/comment all those M73 lines.
Something like this (I have not tested it):

```
#!/usr/bin/python
import sys
import re

settime = re.compile("M73\sP\d+.*")
subst = "; Manually commented: \\g<0>"

out = open(sys.argv[1]+"_notime", 'w')
  with open(sys.argv[1]) as f:
    for r in f:
      if re.search(settime, r) is not None:
        out.write(re.sub(regex, subst, test_str, 0))
      else:
        out.write(r)

```

Noisyfox, 2023-09-15:
Out of curious, why do you want to disable that?

foreachthing, 2023-09-15:
One reason: our Ultimaker 2 does not support it. So, needless gcode is emitted/processed.

quinzico, 2023-10-12:
My Ender 7 has the same issue, its very annoying to hear a bip and see a popup message on the printer screen evey layer.

Nightmare-Faux, 2023-10-24:
I'm using an Artillery X1 and Digant's latest TFT firmware also get an annoying bip and popup every m73 update. Has anyone found a solution to this? For Dummies. I don't know how to use python (or any software for that matter) to post process my g-code.

PR #2940 (2023-12-03): Add option to disable emitting M73 gcode (#2114)
Add option to stop the slicer from emitting any M73 command in gcode, even M73 commands in start/end/custom/etc gcode will be completely removed as well.

Fix #2114
closed 2023-12-03 by SoftFever