OrcaSlicer/OrcaSlicer #2117: Orca Slicer not starting up

mduquette74, 2023-09-15:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Additional system information

CPU: Ryzen 3700x
GPU: 2070 Super
32 GB RAM

### Printer

Tronxy X5SA Pro

### How to reproduce

download and install Orca Slicer
Click to open
Window Pops up in screenshot and will not continue to open.

### Actual results

Window Pops up  and will not continue to open.

**Spiral mode only works when wall loops is 1, support is disabled, top shell layers is 0, sparse infill density is 0 and timelapse type is traditional.
Change these settings automatically? 
Yes - Change these settings and enable spiral/vase mode automatically
No - Cancel enabling spiral mode**

Clicking Yes or No doesn't resolve the issue. Windows just close and nothing happens.

### Expected results

Orca slicer should open and run successfully.

### Project file & Debug log uploads

Spiral mode only works when wall loops is 1, support is disabled, top shell layers is 0, sparse infill density is 0 and timelapse type is traditional.
Change these settings automatically? 
Yes - Change these settings and enable spiral/vase mode automatically
No - Cancel enabling spiral mode

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Akawaki, 2023-10-19:
I also had this problem my fix was similar to chriswilcz. I copied the folder started Orcaslicer so that new config was gernerated and copied all folders. That way I could keep my configs.

bongiben, 2023-10-26:
also having same issue

michael-zock, 2023-12-12:
Wonder if this is the same problem I have on my system (Windows 11 Pro)

1. I've installed 1.7.7 via the EXE, get stuck at the "Loading configuration" part of the splash screen and have to kill the process after 15 minutes without change.
2. Uninstalled it, tried a few days later with 1.8.1, same problem.
3. Removed it and deleted the AppData folders.
4. Tried the portable version and still no change.
5. Even risked starting it as administrator, no change.

Even 8 minutes later, this is all that's in the log file:
```
2023-12-12 23:10:37.217895[Thread 0x00002ea8]:gui mode, Current OrcaSlicer Version 1.8.1
2023-12-12 23:10:37.220894[Thread 0x00002ea8]:Slic3r::GUI::GUI_App::load_language: language , initial: 1
2023-12-12 23:10:37.220894[Thread 0x00002ea8]:language provided by OrcaSlicer.conf: en_GB
2023-12-12 23:10:37.338050[Thread 0x00002ea8]:Slic3r::GUI::GUI_App::Update_dark_mode_flag: switch the current dark mode status to 0 
2023-12-12 23:10:37.338050[Thread 0x00002ea8]:last app version {1.7.7}, current version {1.7.7}
2023-12-12 23:10:37.339052[Thread 0x00002ea8]:begin to show the splash screen...
2023-12-12 23:10:37.392594[Thread 0x00002ea8]:loading systen presets...
2023-12-12 23:10:37.394604[Thread 0x00002ea8]:[BBL Updater]:Checking whether the profile from resource is newer
2023-12-12 23:10:37.394604[Thread 0x00002ea8]:Installing 1 bundles from resources ...
2023-12-12 23:10:37.418655[Thread 0x00002ea8]:[BBL Updater]:Performing 4 updates
2023-12-12 23:10:37.419602[Thread 0x00002ea8]:	Update(C:\Program Files\OrcaSlicer\resources\profiles\Custom.json -> C:\Users\michael\AppData\Roaming\OrcaSlicer\system\Custom.json)
2023-12-12 23:10:37.430124[Thread 0x00002ea8]:	Update(C:\Program Files\OrcaSlicer\resources\profiles\Custom\process -> C:\Users\michael\AppData\Roaming\OrcaSlicer\system\Custom\process)
2023-12-12 23:10:37.430124[Thread 0x00002ea8]:copy_directory_recursively "C:\Program Files\OrcaSlicer\resources\profiles\Custom\process" -> "C:\Users\michael\AppData\Roaming\OrcaSlicer\system\Custom\process"
2023-12-12 23:10:37.535718[Thread 0x00002ea8]:	Update(C:\Program Files\OrcaSlicer\resources\profiles\Custom\filament -> C:\Users\michael\AppData\Roaming\OrcaSlicer\system\Custom\filament)
2023-12-12 23:10:37.535718[Thread 0x00002ea8]:copy_directory_recursively "C:\Program Files\OrcaSlicer\resources\profiles\Custom\filament" -> "C:\Users\michael\AppData\Roaming\OrcaSlicer\system\Custom\filament"
2023-12-12 23:10:37.639258[Thread 0x00002ea8]:	Update(C:\Program Files\OrcaSlicer\resources\profiles\Custom\machine -> C:\Users\michael\AppData\Roaming\OrcaSlicer\system\Custom\machine)
2023-12-12 23:10:37.639258[Thread 0x00002ea8]:copy_directory_recursively "C:\Program Files\OrcaSlicer\resources\profiles\Custom\machine" -> "C:\Users\michael\AppData\Roaming\OrcaSlicer\system\Custom\machine"
2023-12-12 23:10:37.685779[Thread 0x00002ea8]:Slic3r::NetworkAgent::initialize_network_module, can not Load Library for C:\Users\michael\AppData\Roaming\OrcaSlicer\plugins/bambu_networking.dll
2023-12-12 23:10:37.685779[Thread 0x00002ea8]:Slic3r::GUI::GUI_App::on_init_network: on_init_network, load dll failed
2023-12-12 23:10:37.685779[Thread 0x00002ea8]:Slic3r::NetworkAgent::unload_network_module, network module 0000000000000000
2023-12-12 23:10:37.685779[Thread 0x00002ea8]:on_init_network, unload_network_module, result = 0
2023-12-12 23:10:37.685779[Thread 0x00002ea8]:preset_folder: set to empty
2023-12-12 23:10:37.686776[Thread 0x00002ea8]:Slic3r::PresetBundle::load_presets enter, substitution_rule 3, preferred printer_model_id 
2023-12-12 23:10:37.687777[Thread 0x00002ea8]:Slic3r::PresetBundle::load_vendor_configs_from_json: parse C:\Users\michael\AppData\Roaming\OrcaSlicer\system/Custom.json, got description:  "My configurations"
2023-12-12 23:10:37.709289[Thread 0x00002ea8]:update_compatibility for all presets enter
2023-12-12 23:10:37.709289[Thread 0x00002ea8]:update_compatibility for all presets exit
2023-12-12 23:10:37.709289[Thread 0x00002ea8]:update_compatibility for all presets enter
2023-12-12 23:10:37.709289[Thread 0x00002ea8]:update_compatibility for all presets exit
2023-12-12 23:10:37.709289[Thread 0x00002ea8]:update_compatibility for all presets enter
2023-12-12 23:10:37.709289[Thread 0x00002ea8]:update_compatibility for all presets exit
2023-12-12 23:10:37.709289[Thread 0x00002ea8]:Slic3r::PresetBundle::load_presets finished, returned substitutions 0
2023-12-12 23:10:37.709289[Thread 0x00002ea8]:create the main window
2023-12-12 23:10:37.959775[Thread 0x00002ea8]:WebView::CreateWebView: file:C:/Program%20Files/OrcaSlicer/resources/web/h
```

Kazkondu, 2024-02-08:
I have the same problem after flow calibration

MustangPeter, 2024-02-10:
> I have the same problem after flow calibration

It all looks like Im facing the same problem.

Been using OrcaSlicer for a while without any problems, but since I did a max flow test Im facing the same problem

Removing all Folders (but leaving OrcaSlicer.conf and OrcaSlicer.conf.bak) from ".\Users\[user]\AppData\Roaming\OrcaSlicer", starting up Slicer, exiting Slicer, moving Folder files back and its working just fine for an undefined time (could be days, hours, weeks, ...).

HaernCalistinson, 2024-02-13:
I just had this happen to me tonight and I narrowed it down to whatever profile you had active when running the flow tuning. If you go into /AppData/Roaming/OrcaSlicer/user/[numbers]/process and delete the profile (and associated info file) for the profile you had active. If you back the file up, you can use the settings from the JSON to recreate a new profile. Just be aware that your wall loops, top, and bottom layers may still be set to spiral settings (1 wall, 0 top, 0 bottom) in the json.

DanRRamey, 2024-02-19:
If yoiu have this issue, just need to do the following:
1. Go to  /AppData/Roaming/OrcaSlicer/user/[numbers]/process
2. Open the .JSON file for the profile you want to open. 
3. Do a search for "spiral_mode", Change it from a 1 to a 0 and save the file. 
4. Start Orca Slicer and all is good to go. 
No need to delete the Configuration file.

CTechLite, 2024-03-20:
> @DanRRamey i just tried your method to no avail. also spiral mode was already set to zero. saved it am still getting the error message for the spiral mode. so im not real sure what to do at this point. please advise .

You may need to find the specific profile that was being used the last time Orca worked. For me there were several Json files and some were nested in the Base folder.

Walrus68, 2024-03-23:
I just had this happen and tried an update to 1.9.1 but no change. I managed to open OrcaSlicer after editing OrcaSlicer.conf file in Appdata\Roaming. I had been working with presets for ABS GF and in the config I changed:
"presets": {
        "filaments": null,
        "machine": "MAX .6 ABS GF"
    },

to->

"presets": {
        "filaments": null,
        "machine": "MAX .6 Easy Nylon"
    },

I also changed recent last_opened_folder and recent projects in case they had the problem and were pre loaded on startup.

Slicer now opens and i touched a few "vase" related settings and closed the app and reopened fine.

marcofailli, 2024-05-13:
Hi everyone,
I'm facing the same issue after installing Orca and it does not get fixed by deleting the App data folder you indicated...
Any suggestions?

TuTanK5, 2024-06-03:
I'm having the same issue. I tried all the suggested changes but nothing works.

marcosikkens, 2024-06-16:
> I'm having the same issue. I tried all the suggested changes but nothing works.
> 
> Edit 1: I think my problem is with AMD Aldrenalin. My CPU: Ryzen 5600G, GPU: RX 6650 XT. I uninstall Aldrenalin and update graphic driver to latest version (15May), and Orca works now. If I have Aldrenalin it doesn't work regardless of driver.
> 
> PS: I also had a bunch other problems with Aldrenalin. It's the worst.

I have the same problem, i'll try updating the driver from 13-3 to 15-5 version.

a-b, 2024-06-28:
In a similar situation, 2.1.1 got stuck at the "Loading configuration" on macOS Sonoma 14.6.
I was able to "fix" it by renaming the config folder:

```sh
cd '/Users/ab/Library/Application Support'
mv OrcaSlicer OrcaSlicer.bak
```

As result I lost my config, but at least app is loading.

TuTanK5, 2024-07-22:
> > I'm having the same issue. I tried all the suggested changes but nothing works.
> > Edit 1: I think my problem is with AMD Aldrenalin. My CPU: Ryzen 5600G, GPU: RX 6650 XT. I uninstall Aldrenalin and update graphic driver to latest version (15May), and Orca works now. If I have Aldrenalin it doesn't work regardless of driver.
> > PS: I also had a bunch other problems with Aldrenalin. It's the worst.
> 
> I have the same problem, i'll try updating the driver from 13-3 to 15-5 version.
> 
> apparently also bambu studio has the same issue.
> 
> [TuTanK5](https://github.com/TuTanK5) where did you get the driver?

I used the AMD Software: PRO Edition from AMD site but choose to install driver only instead of full installation.

marcosikkens, 2024-07-22:
Thanks for the info, for me after updating to the latest version at that time fixed it for me. Strange behaviour..

closed 2024-10-28 by github-actions[bot]

AlexanderPoe, 2025-02-17:
Same issue Orcaslicer 2.2.0

tried everything on this page to no avail, but what provokes it is maybe Bambu Network Plug-in. 

To recreate the steps: erase inside 'AppData/Roaming/OrcaSlicer', open orca, it asks to install network pug-in, install, it asks to download users presets from the cloud, after it does the same error and the same quit on start up issue.

if just clean install network plugin - no issue.