OrcaSlicer/OrcaSlicer #2120: Allow non-BambuLab printers to have custom sheet profiles

R4toLoko, 2023-09-15:
**The problem**
**TLDR: Sheets have different offset and require to have their own profile forcing us to have many printers profile for each sheet..**

At the moment, only BambuLab printers have sheet profiles
[image]

[image]

But other printers don't use those sticker sheets that requires different temp, but we do need sheet profiles to set the offset.

[image]

[image]

-----
**Which printers will be beneficial to this feature**
I believe that any non-BambuLab printer running Klipper will benefit from this.
Prusa manages sheet within the printer firmware and Marlin prints won't benefit from this.

-----
**Describe the solution you'd like**
**SIMPLE**

Have sheet profiles enabled so we can update our MACROS to use the var `curr_bed_type`
Not just that but allow us to add sheet profiles just like we do with printer and filament.

Having High Temp Plate profile for example will make no sense at all but be able to allow us to custom the field with like Textured, Smooth, PEO, PEY, any other name.
This will remove any complexity from you guys and give us the freedom to create a sheet with the name that work for us.

-----
**Describe alternatives you've considered**
**ADVANCED**

On my Voron, with 5 different sheets with their unique offsets, I have this start g-code.

[image]

On kllipper `PRINT_START` I have it set like so:

```
{% set z_offset = params.ZOFFSET|default(0.000)|float %}

{% if z_offset != 0 %}
     SET_GCODE_OFFSET Z_ADJUST={ z_offset } MOVE=1; SET Z-OFFSET
    {% endif %}
```
and `PRINT_STOP` to clean the off-set in case I change the sheet for the next print, otherwise, klipper will add any new offset on top of the current offset and poop will go down. 
Ask me how I found that out the hard way

[image]

So the alternative would be:

1. have the sheet profiles enabled
2. add a sheet off-set field
3. **IF OFF-SET FIELD IS ENABLED ONLY,** it should require an off-set value and it must be passed like above as part of the `PRINT_START` on the same line at the end of it, not before and not after or it will give `must home first error` if set before or it will dig the sheet with the nozzle if done after and the sheet is higher than 0.000

If I have 3 klipper printers, I won't need to manually change their macros, all of them will understand `SET_GCODE_OFFSET Z_ADJUST=-0.105 MOVE=1` sent from OrcaSlicer and will immediately use the new off-set 

-----

kylek29, 2023-09-15:
This would be beneficial to BambuLab owners as well because from what I can tell, that dropdown just sets the curr_bed_type value and the Machine Start G-Code does an if statement to set the Z-offset.

Certain 3rd party bed plates are thicker and/or have a transparent layer that mess with my X1C (Whambam). Be nice to have a "Whambam" dropdown.

R4toLoko, 2023-09-16:
> This would be beneficial to BambuLab owners as well because from what I can tell, that dropdown just sets the curr_bed_type value and the Machine Start G-Code does an if statement to set the [...]

Awesome, the more printer benefiting from this, the more likely to be implemented.

Thank for the input.

dJOS1475, 2023-10-23:
I would very much like this feature too.

SoftFever, 2023-10-23:
I intentionally disabled it because I personally find it not very useful and prone to errors. 
There have been many instances when I forgot to set the correct bed type, which resulted in the wrong bed temperature being set

dJOS1475, 2023-10-23:
> I intentionally disabled it because I personally find it not very useful and prone to errors. There have been many instances when I forgot to set the correct bed type, which resulted in the wrong bed temperature being set

My use case is simply to set z offset for textured vs smooth. Because I use a Super Pinda inductive probe, I need to lower the offset by -0.275 when I switch to the powder coated plate. I’d love to automate this instead of remembering to apply my presets in Klipper.

R4toLoko, 2023-11-08:
> I intentionally disabled it because I personally find it not very useful and prone to errors. There have been many instances when I forgot to set the correct bed type, which resulted in the wrong bed temperature being set

That bed temperature is restricted exclusively to BambuLab printers so disabling it because of a single printer, is automatically punishing everybody else who doesn't own BambuLab printers :)

Like @dJOS1475 , my use case like many folks is explicitly for z-offset.

- LDO Textured: -1.1
- PEO Smooth:  -0.160
- PEO Textured: -0.18
- PEY Smooth: 0.15
- PEI Smooth: 0.04 
- PEI Textured: -0.220

[image]

Because I cannot use the bed type, I have to have a ton of Printer profile for each sheet which I have forgotten to change it in the past and well, the sheet got a kiss from the nozzle.

The only way I found to automate this because hell no changing the offset manually everytime, each printer profile has this vars with its offset:

[image]

macros.cfg

```
[gcode_macro PRINT_START]

    {% set z_offset = params.ZOFFSET|default(0.000)|float %}

    {% if z_offset != 0 %}
     SET_GCODE_OFFSET Z_ADJUST={ z_offset } MOVE=1; SET Z-OFFSET RIGHT AWAY
    {% endif %}
```

Klipper doesn't clean offset automatically.
[image]

That way I "just" need to remember to pick the right printer profile with the right sheet name, and klipper takes care of the rest.
If we could use the bed type, perhaps the management would be easier with a single printer profile and bed sheets.

-----

We all know that the profiles are locked, any filament and speed profile I have is locked to this printer. 
If I get a 2.4 tomorrow, I will need to create another 10 printer profiles.

Luckily, I just "saved as" so I can still access the filament and speed profiles no matter the printer profile I get.
That is also a locker that should be removed, it adds non-necessary overhead to manage profiles.
I have tried to change that in the past by editing the JSON files but this locker shouldn't exist and the change has to be done on the slicer itself.

Take PrusaSlicer as example, the sheets profiles are managed by the printer firmware so non-Prusa printers have to either make manual changes or do what I did above. Prusa is an Apple so don't expect that to change.
With OrcaSlicer, the feature is right there, locked haha

But it does not matter if I am using the stock printer profile or my profile, I can still access the filament and speed profiles as you would expect.
IMO we have a 2-in-1 problem.

[image]

[image]

HakunMatat4, 2023-11-29:
@igiannakas @Noisyfox do you think this is a hard implementation or does our pov makes sense of why this restriction should be lifted??

Thank you

amdavidson, 2023-12-11:
I too would like to have this option so that my macros can make adjustments based on bed type. 

I'm not sure I agree with the argument that we can accidentally choose the wrong one as that's basically true of all slicer settings. I would guess that there are not many among us who have not accidentally sliced something as PLA when they meant to choose ABS or some other accidental toggle. 

If you really insist on protecting users from themselves, maybe hide access under a checkbox in settings or in the printer config.

Neiizo, 2023-12-15:
I would also really like to see this option ! The difference in z offset between texture and non texture pei plate is non negligeable.

Gridzilla, 2024-01-04:
Please add this back! Would really love to use custom plates for my Voron.

SoftFever, 2024-01-04:
I have heard your voices.
I will add an option to enable multiple bed types.

SoftFever, 2024-01-05:
Solved in 1.9.0

closed 2024-01-05 by SoftFever

Gridzilla, 2024-03-12:
Broken again in 2.0 beta

https://github.com/SoftFever/OrcaSlicer/commit/eb5a1bc7135462fb7cd04fa0853d08459516a75d

[image]

dJOS1475, 2024-03-13:
Howdy, I'm on 2.0.0-dev and I can't figure out how to configure each bed type on my Voron based printer - am I missing something?