OrcaSlicer/OrcaSlicer #2133: UX glitches on Ubuntu dark mode

guestisp, 2023-09-16:
### OrcaSlicer Version

1.7

### OS version

Ubuntu 23.04

### Printer

BLV mgn Cube

### How to reproduce

On ubuntu in dark mode (don't know if it's the same with light mode), there are some text not visible, like dark gray on black or black on black and so on.

[image]

### Actual results

some texts are not visible

### Expected results

All texts should be visible

### Project file & Debug log uploads

No project as this is not related to a single probject

### Checklist of files to include

- [X] Log file
- [X] Project file

TexZeTech, 2023-09-16:
I believe this is the same as https://github.com/SoftFever/OrcaSlicer/issues/2086

closed 2023-09-18 by guestisp

guestisp, 2023-09-18:
Guys please take a look at this issue, so that i can trash away prusaslicer :-D :-D