OrcaSlicer/OrcaSlicer #2136: Linux -- Project tab - Read only?

TexZeTech, 2023-09-16:
### OrcaSlicer Version

1.7.0

### OS version

MANJARO 6.1.51-1

### Additional system information

Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.7
KDE Frameworks Version: 5.109.0
Qt Version: 5.15.10
Kernel Version: 6.1.51-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-3770K CPU @ 3.50GHz
Memory: 15.6 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080/PCIe/SSE2

### Printer

Custom profile. #Customized Robo3dR1+

### How to reproduce

1. Click on create or Open project.
2. Make multiple plates and import/create objects.
3. Save project
4. Click on the Project tab and try to use it.

### Actual results

Everything seems read-only.

[image]

### Expected results

To use the Project tab as expected.

I will say I opened 1.6.3 and it just shows

**Back up your configs first.**

So I went back in time to[ Orcaslicer 1.5.0 ubuntu app image](https://github.com/SoftFever/OrcaSlicer/releases/download/v1.5.0/OrcaSlicer_V1.5.0_Linux.ubuntu.zip) and got this result on a new project with no saves.

I also tried to add images to the project tab and while it did open the file dialog it didn't seem to import. 

[image]

I then saved it to see if opening it in v1.7.0 would change something... it did change something it changed the screen to this 
[image]
(I'm happy I backed my configs up)
closing and relaunching it had the same result.

after restoring the configs I had and then loading up the model in v1.7.0 I get this result but still cannot edit.
[image]

### Project file & Debug log uploads

[testOS1.5.0.zip](https://github.com/SoftFever/OrcaSlicer/files/12641237/testOS1.5.0.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Desterly, 2023-09-19:
This is an upstream issue.  Bambu Studio 1.6 rendered the project tab unusable and they state that it "needs to be rewritten"

you can see one such reference here: https://github.com/bambulab/BambuStudio/issues/2343

LJKraus, 2023-09-26:
Bambu has re-purposed the Project tab and now the only time it will be populated is with data from a Makerworld print project.  It will not be usable for private use.  If you want to put your data there, you will have to make it public on Makerworld.

TexZeTech, 2023-09-26:
> Bambu has re-purposed the Project tab and now the only time it will be populated is with data from a Makerworld print project. It will not be usable for private use. If you want to put your data there, you will have to make it public on Makerworld.

Where did they mention this change?

LJKraus, 2023-09-27:
https://github.com/bambulab/BambuStudio/issues/2601#issuecomment-1733202269
https://github.com/bambulab/BambuStudio/issues/2602#issuecomment-1733226500

TexZeTech, 2023-10-01:
> [bambulab/BambuStudio#2601 (comment)](https://github.com/bambulab/BambuStudio/issues/2601#issuecomment-1733202269) [bambulab/BambuStudio#2602 (comment)](https://github.com/bambulab/BambuStudio/issues/2602#issuecomment-1733226500)

Well, that sucks.... thanks for the info.

Now to go make a feature request.

closed 2023-10-01 by TexZeTech