OrcaSlicer/OrcaSlicer #2140: Random Layer Lines on Toplayers

ALEX-9-99, 2023-09-16:
### OrcaSlicer Version

1.6.4 - 1.7.0

### OS version

Windows 10

### Printer

Bambu Lab P1P/S

### How to reproduce

1. Get the sample file from Printables - https://www.printables.com/de/model/228249-filament-samples-42-materials
2. import "PLA_Filament Sample.stl" from shared folder into orca slicer
3. Sliced with default P1P settings and Bambu PLA filament (reproducible with other filaments)

### Actual results

There are strange patterns appearing on the surface.
Possibly in combination with the letter placement.
[image]

### Expected results

The Print Surface should vbe even with no strange Pattern or higups
[image]

### Project file & Debug log uploads

[PLA_filament_sample_Report.zip](https://github.com/SoftFever/OrcaSlicer/files/12641493/PLA_filament_sample_Report.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

samwiseg0, 2023-09-18:
Have you played with the one wall threshold? You should be able to lower it and get your desired results. Turning off one wall top layer will achieve the same thing. 
[image]

ALEX-9-99, 2023-09-25:
Thanks for the help and sorry for bothering you.
I've never heared about this setting bevor.

closed 2023-09-25 by ALEX-9-99