OrcaSlicer/OrcaSlicer #2142: VFS could be easily improved for a huge QoL update

thhdragon, 2023-09-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

For example if I use my Sunlu PLA+ Black profile set for 20 VFS at 235C temp. For a model with a lot of small details, low layer height, or one I would like to print slower for quality reasons, the easiest way to slow the print down is to lower the VFS instead of changing all the speed settings. For example if i cap the VFS at 12 I don't need my hotend to be 235C it can be 210c to print 12VFS. But who wants to create multiple filament profiles for the same roll just to change the temp/VFS

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

Should work on any printer OS

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

There could be a section in the filament setting where you can put the max VFS value you can get at the max temp you plan to use. People could then do a VFS test, either the print or the extruder skipping test at lower temperatures to find other temps for other VFS methods. If we find the min/max and then a few values in between, then when for example you do a 0.08mm layer print (just an example number, 12VFS at 0.08mm layer height is like 400mms linear speed. I think it would be very useful if there were maybe a VFS tab in filament settings where we could add VFS/temp numbers we've gathered on our filament and then possibly a checkbox to pick which VFS/temp setting (pictured the best example i could come up with)

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

The other ways I've come up with is keeping them all in the notes and doing it all manually, or creating like 4 filament profiles for the same roll of filament but just the temperature and VFS changed. 

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

thhdragon, 2023-09-17:
Im also not sure but pressure advance might also change

kylek29, 2023-09-19:
This is something I bumped into with PETG a few weeks back since it required significant tuning. Personally, I think we've outgrown the original PrusaSlicer design for having Printer, Filament, Process as our settings base. There's so many settings, filament types, and variables now but to change just a few, we have to fork the main .. except User Presets can't inherit from other User Presets (at least not as far as I've tested), they only inherit from the System based ones (which I assume is because those are Read Only unless you manually edit them).

One tweak I thought about was to add a secondary "modifier" presets that you can stack under each. In this example, we'll layer top to bottom (so the top most is applied, then the second, down the tree branch), so any conflicts are overridden as the lower presets take precedence. These could be one setting or a multitude of settings presets.

Printer: X1C
-- Modifier: Fast Start

Filament: Voxelab PETG+
-- Modifier: VFS Min
-- Modifier: Temperature Min

Process: Normal @BBL 
-- Modifier: Layer 0.1
-- Modifier: Speed 50%

** In this example, I would have configured my presets as individual (or close to it) settings. But users could very well combine these into containing multiple common settings.

closed 2023-12-27 by github-actions[bot]