OrcaSlicer/OrcaSlicer #2144: Fan turning on during purge

pennystockss, 2023-09-17:
### OrcaSlicer Version

1.7

### OS version

Windows 11

### Printer

P1P/P1S

### How to reproduce

Use AMS
Have filaments with different fan speed (i.e. min fan speed or aux fan speed differs between filaments)

### Actual results

The fan kicks on during purge at random fan speeds.

In one test, example, right above T5 I get a M106 S76. It's not always right above the T gcode, depending on if you modified the change filament code. With my previous custom profile, the fan speed would appear a few lines above the T or below. Tested on factory profile and it reproduces.

The code and speed varies slightly. It seems it could be a M106, M106 P1 or M106 P2. With the fan speed being different.

This causes a huge issue as when purging, the filament cools down and does not create a solid blob. It creates strings and can't plop down the chute causing backup.

### Expected results

Fan off during purge.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/12642166/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

pennystockss, 2023-09-25:
Bump. Am I the only one experiencing this issue? I created a cube, painted a side black. 2 different profiles used, Bambu Matte and Bambu Normal. This is the tool change Gcode:

`M620.1 E F548 T240
T1
M620.1 E F523 T240`

In an example attached in the original post, I increased Min Fan Speed to 75, decreased Max Fan Speed to 85 and changed Aux Fan to 25 for 1 of the filaments, I then get fan speeds added during tool changes, majorly messing up reliable purges:

`M620.1 E F548 T240
**M106 S168
M106 P2 S178**
T1
M620.1 E F523 T240`

philltran, 2023-09-30:
I am also experiencing this issue.

pennystockss, 2023-09-30:
Ok good it isn't just me. This needs way more attention because it literally is breaking prints by the purge chute piling up because it's full of long strings instead of blobs that can drop. I bet a lot of people don't even notice it or experience it because their fan speeds are not different between profiles. 

**Also present on Layer Time**. Just printed with supports for PLA. Support had a layer time of 8 seconds, other profiles had layer times of 4 seconds. **This was the only difference.** Checked Gcode and found:

M620.1 E F299 T240
****M106 S155**  <----------FAN SPEED ADDED BEFORE TOOL CHNAGE**
T0 <--------- Tool Change
M620.1 E F523 T240

Then the only change I made was lowering layer time of PLA support to 4 seconds to match the other filament and I get:

M620.1 E F299 T240
T0
M620.1 E F523 T240

Please can this get some attention. It's tiring to have to check every gcode before print because a 24 hour print with several hundred tool changes will fail if the fan is on during purge. @SoftFever

CristianGTR97, 2024-01-02:
I'm having a similar issue.

The configured filament change g-code has the following lines:
;
M400

T[next_extruder]
;

But the generated output g-code when it changes to T1 in the filament change section looks like this:
;
M400

M106 P2 S255
T1
;

Similarly, when it changes to T2 a gcode line is added setting the P2 fan to S178

The configured auxiliary fan speed for the filament 2, which corresponds to T1 is 100% or 255 in 8 bit
But the configured AUX fan speed for the filament 1 (T0) is 70% or 178 in 8 bit

I've noticed that if the settings for the Part Cooling or AUX fan are different, gcode lines are added by the slicer before the toolchange command to set the fan speed to the new filament's fan speed

But when the fan settings are the same, these gcode lines don't get added in the output gcode.

I tested other filament dependent settings, by making them different between the two filaments I was using, to see if any other parameters would also get adjusted as well before the T command. The tested settings were:
-Flow Ratio
-Pressure advance
-Chamber temperature
-Activate temperature control
-Nozzle temperature
-Bed temperature
-Max volumetric Speed
-Part cooling fan speed
-Min print speed
-Auxiliary fan speed
-Activate air filtration
-Exhaust fan speed

The only settings that seem to trigger the introduction of the new gcode lines before the T command, when they are different between filaments, are the part cooling fan speed, and the auxiliary cooling fan speed.

In my opinion, Filament dependent settings should be updated with the filament start gcode, not with the filament change gcode. The only exception being (maybe) the nozzle temperature, since you need it to be the right temperate before you melt the new material.

I'm running Orca Slicer 1.9.0-beta

And I'm using a Bambulab P1S with AMS

closed 2024-04-11 by github-actions[bot]

pennystockss, 2024-04-22:
It still persists… Accidentally selected a default profile for one color last week and nearly wrecked a 1.5day print, made a huge pile, backed up, front cover popped off from hitting it, the usual (thankfully resumed with layer no shift). I guess most people don’t mess with the fan speeds and then they’re always the same. Issue needs to be reopened and looked at, major bug IMO. Maybe I’ll test in Bambu studio, I think it’s present in there and open an issue on their build.