OrcaSlicer/OrcaSlicer #2145: Support top interface printing middle air

R4toLoko, 2023-09-17:
### OrcaSlicer Version

1.7.0

### OS version

Mint Cinnamon 21.1

### Printer

Voron Trident

### How to reproduce

1. Slice anything that requires support forced or not
2. Check the support top interface

### Actual results

I don't use supports so this could be me doing user stuff.
I am testing this Organic tree and the top interface is being printed middle air.

[image]

[image]

If I reduce the top interface spacing, it improves but still prints middle air.

[image]

### Expected results

Top interface layer should be printed like tree does:

[image]

### Project file & Debug log uploads

[master_chief.zip](https://github.com/SoftFever/OrcaSlicer/files/12642282/master_chief.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2023-09-18:
duplicate of this: https://github.com/SoftFever/OrcaSlicer/issues/2038

Same as prusa slicer - bug is carried over: https://github.com/prusa3d/PrusaSlicer/issues/11155

R4toLoko, 2023-09-18:
> duplicate of this: https://github.com/SoftFever/OrcaSlicer/issues/2038
> 
> Same as prusa slicer - bug is carried over: https://github.com/prusa3d/PrusaSlicer/issues/11155
> 
> 

I need to start searching first more before opening a ticket 🤦

closed 2023-09-18 by R4toLoko