OrcaSlicer/OrcaSlicer #2150: Changing the printer setting changes filament and/or process preset

julie777, 2023-09-17:
### OrcaSlicer Version

1.7.0

### OS version

windows 10

### Printer

P1P

### How to reproduce

1. click on the dropdown and select a different printer preset
2. check to see if either the filament or the process preset changed
[image]

select full
[image]
result
[image]

change filament
[image]

now change printer
[image]

### Actual results

it does not work correctly

### Expected results

notes above

### Project file & Debug log uploads

[debug.zip](https://github.com/SoftFever/OrcaSlicer/files/12643722/debug.zip)
While uploading the debug log I noticed that it is full of socket messages. That does not look right either.

### Checklist of files to include

- [X] Log file
- [ ] Project file

samwiseg0, 2023-09-18:
This was actually an intentional change made a few releases ago. 
[image]

rizwansarwar, 2023-09-19:
> This was actually an intentional change made a few releases ago. [image]

So for example if I have a project saved with some settings with a printer selected. If I change to another printer that is a similar printer I need to save the settings again for that printer? Who came up with this genius improvement? Sounds like really badly thought out.

samwiseg0, 2023-09-19:
> Who came up with this genius improvement? Sounds like really badly thought out.

Not sure but having a slightly better attitude about it may get you further and others would be more willing to help and/or make that an optional feature.

kylek29, 2023-09-19:
> So for example if I have a project saved with some settings with a printer selected. If I change to another printer that is a similar printer I need to save the settings again for that printer? Who [...]

From the looks of it, the intention is that they (the people who implemented it) think each individual printer would be unique. They didn't foresee a use case where that would not be the case and a user maybe using the printer dialog as pseudo printers for swapping settings.

bubbleytea, 2023-09-20:
+1 would appreciate a way to toggle the previous behavior. I'm often switching between printer configs but want all the other settings to remain the same, so I was surprised to find that this is the new default.

rizwansarwar, 2023-10-28:
@SoftFever Any love for this unwanted change?

SoftFever, 2023-10-29:
> @SoftFever Any love for this unwanted change?

Can you share use cases why using pseudo printer profiles? It is not designed in that way

rizwansarwar, 2023-10-29:
> > @SoftFever Any love for this unwanted change?
> 
> Can you share use cases why using pseudo printer profiles? It is not designed in that way

@SoftFever Thank you for responding so quickly. I can certainly list the use case that I face but others might be able to add more. 

I have a mix of printers, Bambu, Creality and Prusa. Normally, I create project files and modify settings as needed (some global to profile and some per object). I have all the printers connected via Octoprint or Klipper or Bambucloud, so I send print jobs from Slicer directly to them. 

In my saved project file, at the time of saving, what ever printer was selected is saved (lets assume it was Prusa Printer #1). Next time when I open that project file, Orca will load the project with same settings and same printer. If I want to send to lets say Prusa Printer #2 at this point, Orca will revert to default print profile as soon as I switch printer from Orca. That is not what I want, I just want to send same project to a different printer. Of course I would have a different project file for a Klipper based printer, but for similar type of printers, it should not reset my settings. 

For reference, in Prusa Slicer, Super Slicer, Cura, this issue does not exist, if you change the printer between Prusa printers, it does not reset your project. 

And this was not a problem before 1.7.0 in Orca, it used to work just fine, but since 1.7.0, it was assumed that each printer+profile is unique combination. 

It is also possible that I don't quiet understand how we are supposed to handle the above in Orca, but for someone who deals with multiple printers every day, switching between them for different jobs, it is important to have this sort of control. 

Hope this makes sense and would love to hear your and others feedback on this. Thanks

skynet01, 2024-01-22:
Another add to this is it makes sharing of 3mf files pretty much useless. 
if you download any 3mf file that someone shared for example from Bambu Maker World their profile will be different than yours, when you switch it to your printer all settings they made to make the print correctly will get cleared. At the very least we should get a dialog asking if we want to migrate the settings that were changed when printer profile is switched.

skynet01, 2024-04-22:
Has this bin fixed?

rdbischoff, 2024-06-29:
really would like to be able to open 3MFs and switch to my printer but the color painting stay and then I can switch to my version of the colors I will be using. and be able to keep their settings and only tweak them for what I want to change rather than changing all settings and colors. there is no point in passing around 3MFs without this option to keep their colors then. @SoftFever

skynet01, 2024-09-29:
yes but has this got resolved?

closed 2025-01-06 by github-actions[bot]