OrcaSlicer/OrcaSlicer #2153: 1.7.0 printer kinematics behave different than in 1.6.6

brentrichardjr, 2023-09-17:
### OrcaSlicer Version

1.7.0

### OS version

Windows 11

### Printer

Voxelab Aquila w/ Sprite Extruder on Klipper

### How to reproduce

Upgrade from 1.6.6 to any 1.7.0 release

### Actual results

1.7.0 (beta or release) doesn't behave the same as in 1.6.6 in terms of kinematics. The bed accelerates and decelerates quickly, causing banging noises and layer shifts. This issue previously occurred in the 1.6.4 beta releases but was fixed by release. I initially updated when the 1.7.0 beta was available, but reverted after the issue awaiting full release. It seems like the speed settings (i'm assuming travel speeds) aren't being recognized/followed correctly.

No change in profile. Fresh install with import of profile also didn't work as expected. Reverting back to 1.6.6 fixes the issue.

### Expected results

1.7.0 should behave the same way as 1.6.6 in regards to printer kinematics.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

StudioKris, 2023-09-19:
I've encountered the same issue of layer shifting after upgrading to version 1.7.0. To diagnose the problem, I conducted a Retraction Test, and the shift occurred during the printing of the second layer. Upon further investigation, I identified the problematic lines of gCode in the Retraction Test when the print failed:
`G1 X90.844 Y114.446 E.10387
; stop printing object retraction_tower.stl id:0 copy 0
M106 S255

;LAYER_CHANGE
;Z:0.4
;HEIGHT:0.2
EXCLUDE_OBJECT_END NAME=retraction_tower.stl_id_0_copy_0
;_SET_FAN_SPEED_CHANGING_LAYER
; Calib_Retraction_tower: Z_HEIGHT: 0.4, length:1.5
M140 S60 ; set bed temperature
; printing object retraction_tower.stl id:0 copy 0
SET_VELOCITY_LIMIT ACCEL=7000 ACCEL_TO_DECEL=3500
G1 E-1.5 F1800
EXCLUDE_OBJECT_START NAME=retraction_tower.stl_id_0_copy_0
G1 Z.6 F21000
G1 X127.624 Y116.831 Z.6
G1 Z.4
G1 E1.5 F1800
SET_VELOCITY_LIMIT ACCEL=100 ACCEL_TO_DECEL=50
;TYPE:Inner wall
;WIDTH:0.449999
G1 F8843.492
G1 X126.962 Y116.918 E.02215`

The problematic part in the gCode is the "SET_VELOCITY_LIMIT" command, which sets an acceleration value that is excessively high for my printer to handle. Based on my examination of the OrcaSlicer source code, it appears that this acceleration value corresponds to the travel acceleration.

In the OrcaSlicer printer settings, the "Maximum acceleration for travel" is configured at 9000 mm/s^2. However, my printer's maximum supported acceleration is only 2500 mm/s^2, as exceeding this value leads to missed steps. Unfortunately, I cannot manually adjust this value in the UI because the field is disabled.

Furthermore, it's concerning that a slicer utilizes the "SET_VELOCITY_LIMIT" command, as this allows the slicer to override the acceleration limits specified in the printer's firmware. This poses a risk, as unadapted acceleration values set by the slicer could potentially damage the hardware.

samwiseg0, 2023-09-19:
@StudioKris Could you by chance share your project?

Also do you have acceleration control enabled at all? 

With it disabled I sliced an object and did not get any erroneous `SET_VELOCITY_LIMIT` in my gcode. I am just wondering if this is a settings issue vs a bug?

[image]

samwiseg0, 2023-09-19:
I do not disagree. It is not the most intuitive to place a `0` there to "disable" it. 

@brentrichardjr Can you provide your project file? Can you provide more details of what is different in your gcode between the 2 versions? That will help get to the bottom of your issue.

brentrichardjr, 2023-09-19:
I did a side by side of v1.6.6 and v1.7.0. I did notice that v1.7.0 added a Maximum Acceleration for Travel to the Machine Settings. It's greyed out and set at 9000. If its using this over my settings, than that would fit with what the banging on travel movements.
[image]
[image]

This is what I was slicing/printing when I did the update. 
https://www.printables.com/model/511962-taurus-v4-cooling-duct-for-creality-sprite-extrude
I successfully printed it using 1.6.6. I've zipped it saved as both the sliced gcode as well as the project file.
[Project and Gcode.zip](https://github.com/SoftFever/OrcaSlicer/files/12666992/Project.and.Gcode.zip)

samwiseg0, 2023-09-19:
@brentrichardjr based off the project you provided you have acceleration control turned on and your travel is set to 7k. 
[image]

I actually suspect there was a bug and now it is fixed as of 1.7.0. The reason being is that you had acceleration control **_turned on_** for your project while using v1.6.6 _but_ it was not actually being emitted to gcode. Note how find only found a single instance of setting the acceleration. This was the bug. 
v1.6.6 Gcode:
[image]

v1.7.0 Gcode:
[image]

So in summary. Your accelerations were enabled from the start and were not properly configured for your printer but it did not cause any ill effects because it was never emitted to gcode. After you updated to v1.7.0 the acceleration commands were actually emitted to gcode causing you issues. 

To "fix" this you will want to either disable acceleration control under "speed" or properly configure it for your machines limits. 
[image]

As far as the screenshot you provided. Those are machine limits and are not used for anything but estimating how long it will take to finish the print.

brentrichardjr, 2023-09-19:
Thanks for looking into this. I'll just calibrate the speeds again now that I know what to expect. I appreciate you.

closed 2023-09-19 by brentrichardjr

samwiseg0, 2023-09-19:
Not a problem. Glad I could help!