OrcaSlicer/OrcaSlicer #2156: Obnoxious message box does not close by itself

julie777, 2023-09-18:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Printer

P1P

### How to reproduce

1. go to the printer and unload filament
2. load new filament
3. go back to PC and look at Orca Slicer 

### Actual results

This message box is showing.

[image]

### Expected results

I have already completed the steps to change filament at the machine. This should have gone away by itself. This has to be the most annoying thing in 1.7.0!

### Project file & Debug log uploads

Why is this required? 
For many issues I do not need to upload logs.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kylek29, 2023-09-20:
Have you tested with BambuStudio to see if it is the same? I noticed these messages (there's a few of them) persist on the printer itself, it's rather annoying, as you have to clear 10 old messages that have stacked up. It's possible Orca is just relaying the printers message queue (would make this a firmware issue).

closed 2023-12-27 by github-actions[bot]

tlhintoq, 2023-12-27:
Also to include in those non-closing messages that come over and over...
> You're in object sequence.  We recommend you auto-layout to avoid conflict.

Got ya, I'll lay out my job, thanks.  Don't need this all the time.