OrcaSlicer/OrcaSlicer #2160: [Feature Request] Anycubic Predator Profile

Frontier314, 2023-09-18:
**Is your feature request related to a problem? Please describe.**
Printer profile for Anycubic Predator.

**Which printers will be beneficial to this feature**
Anycubic Predator.

**Describe the solution you'd like**
I would like to use Orca Slicer to slice files for the Anycubic Predator FDM printer.

**Describe alternatives you've considered**

**Additional context**

rtrevino70, 2023-10-07:
x2 on this feature. I have 3 predators still going strong

djseraph, 2023-10-18:
Are you interested to Machine g-code?

> **La tua richiesta di funzionalità è correlata a un problema? Per favore descrivi. ** Profilo stampante per Anycubic Predator.
> 
> **Quali stampanti trarranno vantaggio da questa funzionalità** Anycubic Predator.
> 
> **Descrivi la soluzione che desideri** Vorrei utilizzare Orca Slicer per suddividere i file per la stampante FDM Anycubic Predator.
> 
> **Descrivi le alternative che hai preso in considerazione**
> 
> **Contesto aggiuntivo**

> x2 su questa funzione. Ho 3 predatori ancora forti

closed 2024-01-24 by github-actions[bot]