OrcaSlicer/OrcaSlicer #2162: X/Y Support Distance Priority

movinimage, 2023-09-18:
When printing with soluble support is is important to use 0 as your z support distance  and x/y support distance to get good quality suppported surfaces. But, when printing using soluble support _only_ for interface layers, the x/y clearance still must be maintained so that the support structure doesnt touch the model. In this case, if an arch or filet is being supported, the x/y distance will keep the curved surface from being properly supported. Cura (and I *think* Prusa) has an option to override this special case called "Z overrides X/Y."

In OrcaSlicer (and Bambu Studio) there is no way to override the X/Y Support Distance in cases where curved surfaces need to be supported. This makes it impossible to properly support a filet or other arch while also maintaining an X/Y support distance for the rest of the print. 

**In Cura, you can see the behavior when "Z overrides x/y" is selected. The fillet is properly supported.**

[image]

In OrcaSlicer, without this option the filet is not supported. In order to support the fillet, the x/y distance must be reduced to 0, in which case the support infill will also bond to the verticals of the part, making soluble support interfaces impossible to use. Instead the entire support must be soluble which results in wasting expensive support filament and requiring an order of magnitude more filament changes.
[image]

movinimage, 2023-10-04:
Just curious if this can be added?

movinimage, 2023-10-05:
I've gotten a few private messages about this post. People are confused. I think I wasn't clear enough..so I hope this helps:

The problem is on the top steep part, not the lower part which is below 45deg. Cura has an option “Z overrides x/y.” for this exact case. it allows the support interface to actually touch the steep overhang even if it is within the horizontal separation distance. The problem is in the second example, the OrcaSlicer example…where the yellow portion (the dissolvable support interface) does not actually touch the model as it must when we are getting up to 65-75-85 deg overhangs.

In OrcaSlicer when using a soluble support interface, with a regular support structure, you have only two options to get curved overhangs properly supported. Both are bad options, that don't solve the problem.

1- If you simply set the horizontal offset to 0 (which is the only option in Bambu Studio), then the entire support structure will touch the model, including the non-dissolvable part, and the model will be ruined.

2- Or, set a horizontal offset …like .35 which I think is default…then, the non-dissolvable part of the support will not touch the model, which is great…but the dissolvable portion of the support wont actually provide any direct support.

The only options you are left with in OrcaSlicer then are to use 100% dissolvable support, with a 0 horizontal offset (which results in astronomic print times and waste because of filament changes)…or model your own supports as separate parts in CAD, and then use the dissolvable filament for them.

I tried out the option in Cura…and it fixes the entire problem with 1 click.

I hope this is clearer.

movinimage, 2023-10-06:
I can see my screenshot is not from an angle that is clear. The yellow does not touch model, and the interface layers are not printed in the air. When fully zoomed in, Orca does not allow smooth small-step rotation of the view, so it is quite difficult to get a good screenshot. 

All3DP gives what might be a clearer explanation of the problem and why the setting in Cura exists. Scroll to "Distance Priority"

https://all3dp.com/2/cura-support-settings-optimize-your-supports/

The need for the override setting is a known issue on all Cura/Slicr derivatives, and is the reason the "Z ovverrides X/Y" setting exists in Cura and in Prusa I believe. I have confirmed it on screen and in printing. The only solution I've found that works is to hand-model supports in CAD and then choose dif. filaments for them in Orca..then do a ton of customizing of settings to get a 'part' to print like a support (interface layers, sparse internals, no walls, etc...).

Here is a better screenshot so that you can see how it actually prints in Orca. 
[image]

Without this setting available, soluble (and sometimes breakaway) support simply doesn't work in many situations unless the entire support structure is printed in the separate support material. This is not practical with a single extruder and filament changes. Best practice requires using support interfaces in the soluble/breakaway material and the rest of the support structure in the original material. This doesn't work though because the support body ends up touching the model. Only "Z overrides X/Y" fixes this issue by allowing the support body to keep a safe distance while at the same time allowing the interface layers to make full z contact with the model to be supported.

Please consider bringing this setting to OrcaSlicer from Cura.

movinimage, 2023-10-06:
Another...maybe clearer explanation than mine:

See here and scroll to "change support distance priority"

https://3dprinterly.com/how-to-fix-cura-supports-not-touching-model-or-print/#:~:text=Another%20strong%20recommendation%20from%20Cura%20users%20is%20to,affecting%20the%20set%20Z%20distance%20to%20the%20overhang.

closed 2024-01-12 by github-actions[bot]

cbaugher, 2024-04-18:
X/Y separation distance needs to be a feature.  I have a situation where it needs to be set further away because the support is sticking to the side of the part.  This was already in PrusaSlicer, so why was it removed in Orca?

movinimage, 2024-04-18:
This topic is specifically about "X/Y Support Distance **Priority**" which is different than X/Y distance. It is about allowing the need for direct Z contact with soluble/breakaway support override the X/Y support distance. Just wanted to make sure you get your quesion in the right place so that it is seen.

cbaugher, 2024-04-22:
> This topic is specifically about "X/Y Support Distance **Priority**" which is different than X/Y distance. It is about allowing the need for direct Z contact with soluble/breakaway support override [...]

Well, it turns out the setting I was looking for does exist.  Being used to Prusaslicer I was looking for a different name and just missed it.  So, please disregard my earlier message.  :-)

wwmotorsport, 2024-07-30:
I'm new to Orca since the tool changer support is being developed, but I primarily print with PETG for a model material and PLA for a support interface layer. Everything about Orca seems superior to Cura, except this one thing... Just like the OP, even if I set a 0 support vertical distance, a non-zero support/object x/y distance results in effectively no support for the part. 

[image]