OrcaSlicer/OrcaSlicer #2170: Make hidden options in search result more clear

Noisyfox, 2023-09-19:
Choose any option that could be hidden under certain conditions, for example the "Overhang speed: Classic mode" toggle is only displayed if "Slow down for overhangs" is turned on:
[image]

Then turn "Slow down for overhangs" off, which should hide the "Classic mode" option.

Then search "Classic mode", which shows this option:
[image]

But if you click it, the app switches to the "Speed" panel but doesn't show the target option, which is confusing.

We should somehow tell the user that this option is currently hidden/disabled/unapplicable in search result, for example by adding a "(Disabled)" prefix/suffix to the result.

I don't want to hide the option from the result, because when you are using search, it's better to tell you that we have that option but it's not usable at the moment and you might need to do some other things first to enable it, other than just hide it which doesn't give user any meaningful information.

kylek29, 2023-09-19:
Yeah, I bumped into this as well and thought I was going nuts not seeing an option (didn't realize it was hidden by another).

Maybe a solution would be to not hide the option, but gray it out as disabled (bonus points for updating the mouseover tooltip to say what disabled it)? That's a more common way of handling UI elements, since shifting / hiding things can confuse users.

samwiseg0, 2023-09-19:
I vote for it being greyed out and not hidden. I find that hidden settings are less user friendly in general and you end up in a where's Waldo situation? That would also users to still be able to search for the feature so they know its there. 

Another Idea I had is that if you try to enable/click on a disabled setting it would tell you what setting would need to be enabled to allow it to be used. Esentually provide information to the user of the feature that supersedes it.

mrelwood, 2023-09-19:
I don’t know how many Prusa based settings are not visible in Orca at all, but I have a hunch that it’s too many to make graying them out applicable. Some of them aren’t in use in Orca under any situation.

I think a good one might be if `Disabled (?)` was displayed in the search results with either a mouse hover tooltip or a popup after a click saying that the visibility depends on other settings, or if it is replaced by other settings altogether. And clicking the result wouldn’t do anything else, not even close the search results. That way the user can easily revise the search.

Ocraftyone, 2023-10-14:
The hard part about this is that the option is not coded in a way that it knows it is dependent on another option, rather there is a method run whenever the config is updated and in there, you manually check if an item is toggled, then hide/reveal manually. To implement this, it would probably be a rehaul of this system by having a callback function on the options themselves.

Myoporum, 2025-02-28:
Needs bumping, as hiding options, rather than disabling them, confuses users who may never know that option is even possible depending on what config they start with.

Ocraftyone, 2026-08-30:
Implemented in https://github.com/OrcaSlicer/OrcaSlicer/pull/12381

MxBrnr, 2026-09-02:
> Implemented in [#12381](https://github.com/OrcaSlicer/OrcaSlicer/pull/12381)

Useful upgrade, but I'd still like to see the settings themselves always visible. I would not consider this PR a fix. Reason being that there are many settings people don't even know exist, so they won't even know to search for them, making that PR change less than half-effective. The better option would be to reveal all settings, and list in the tooltips why the setting might be "disabled." This is exactly what Cura does with their UI, and it is far more effective.

Noisyfox, 2026-10-20:
I'll consider adding a separate view mode (basic/advanced/expert/custom) which user can set which option to show or hide. In the custom mode the option will not be hidden conditionally, but be disabled instead.

closed 2026-10-26 by SoftFever