OrcaSlicer/OrcaSlicer #2172: Add a “Ignore/Emit in gcode” function in Printer setting/Motion ability

jiiimi, 2023-09-19:
**Is your feature request related to a problem? Please describe.**
I still have to edit my gcode manually to remove these lines:
; EXECUTABLE_BLOCK_START
M73 P0 R23
M201 X0 Y0 Z0 E0
M203 X0 Y0 Z0 E0
M204 P0 R0 T0
M205 X0.00 Y0.00 Z0.00 E0.00 ; sets the jerk limits, mm/sec

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
Add in Printer setting/Motion ability a function "Ignore / Emit in gcode" which allows you to choose if you want to customize Motion ability or if you want to use the default values of the firmware.
Or
By setting all values to zero this is not taken into account in the gcode (this is not currently the case)
[image]

Duncro, 2023-09-23:
Yeah, overrides whatever I have on the printer. Biggest annoyance is when I fiddle with settings mid print on the printer, and I click save, then it saves these settings overwriting what I had on the printer (forcing met to start from scratch, making orca slicer a complete no go for tuning your printer and usually forcing you to start from the beginning). These should not be forced slicer specific settings and you should at least have the option to omit this from gcode automatically.

closed 2023-12-31 by github-actions[bot]

Dosclic98, 2023-12-31:
Yeah I think it would be usefull!

jbjhjm, 2024-06-20:
oh how I hate that stupid stale github bot. This is an important bit missing for everyone who doesnt want to have the slicer mess up printer configuration.