OrcaSlicer/OrcaSlicer #2175: Infill Combination ignores Layer Height limits

ZedOud, 2023-09-19:
### OrcaSlicer Version

1.7.0

### OS version

macOS 13.5.2

### Printer

FLSun v400

### How to reproduce

Set:
`Quality -> Layer height` to 0.1mm
`Strength -> Advanced -> Infill combination` is enabled
`Printer Settings -> Extruder -> Layer Height limits -> Max` is less than the `Nozzle Diameter` (as I assume the latter is the calculated limit). For example, 0.3mm

### Actual results

When slicing an object, checking the `Layer Height` view shows us 0.1mm perimeter height and 0.4mm infill height, the latter of which is well beyond the `Layer Height limit`
[image]

I believe this is related to issues #1125 and #2158 

### Expected results

The infill layer height as a result of `Infill combination` should be limited to the `Layer Height limit`, in this case 0.3mm, not 0.4mm

### Project file & Debug log uploads

[InfillCombinationIssue.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12667333/InfillCombinationIssue.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

closed 2023-12-27 by github-actions[bot]

Vorta, 2024-04-12:
Too bad that this was not resolved. I cannot use infill combination in my prints because of this. I have max layer height set to 0.2mm on my 0.4mm nozzle and infill combination still wants to print 0.4mm high infills.

Gamerou, 2024-05-22:
> Too bad that this was not resolved. I cannot use infill combination in my prints because of this. I have max layer height set to 0.2mm on my 0.4mm nozzle and infill combination still wants to print 0.4mm high infills.

same