OrcaSlicer/OrcaSlicer #2178: Support base overlaps causing fist layer collison

Edzi11, 2023-09-20:
### OrcaSlicer Version

1.7.0

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. load model
2. enable support
3. support first layer overlaps

### Actual results

first layer of supports overlap with each other causing collision. view attached image 
[image]
 

### Expected results

when generated the supports first layer chould merge together rather than collide with each other. almost seems like theres a raft being printed even though that isnt enabled. 

### Project file & Debug log uploads

[elder_wand project.zip](https://github.com/SoftFever/OrcaSlicer/files/12668148/elder_wand.project.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

closed 2023-12-27 by github-actions[bot]

ohmdelta, 2024-01-06:
this seems to still be in the latest 1.9.0 release (can also be duplicated in Linux image with close objects)

Steak77, 2025-01-25:
Running 2.1.1 and the issue is still present:

[image]