OrcaSlicer/OrcaSlicer #2181: Repetier G-Code flavor -- supported by original PrusaSlicer & SuperSlicer

Lilith8x, 2023-09-20:
**Is your feature request related to a problem? Please describe.**
Tried running my custom printer with this software, but realised that there is major ... drawback. Tried the speed settings to get proper printing speeds but printer just ... seems to stop and think from time to time - causing blobs everywhere it stops. Im running my printer with Repetier-server, and by following the command terminal I realised the problem:

Marlin uses M205 to send acceleration values - Repetier uses M205 to spit out EEPROM settings. Every time slicer intends to change accelerations, it reports EEPROM status - which stops the print...

**Which printers will be beneficial to this feature**
Any older printers running repetier firmware.

**Describe the solution you'd like**
Repetier flavor of the G-Code - most likely just the speed/junction G-codes are affected.

**Describe alternatives you've considered**
At first i thought I would just add a regex to delete every M205 - didnt exist in OrcaSlicer, but exists in PrusaSlicer&SuperSlicer.

Then I thought about writing a python script to rewrite the g-code - but that is an ugly solution.

**Additional context**
--

hcoohb, 2023-10-01:
I would also love to have a repetier flavor G-code option for my printer as I believe it is the only thing that it supports...

Lilith8x, 2023-10-01:
I wrote a small configuration change to Repetier-Server which seems to work rather well - atleast with my printer...
3x Regex replacements which convert the acceleration/jerk from Marlin to Repetier - use at your own risk.

Repetier-Server-->Printer settings --> G-Codes --> Replacements

Regex 1:
**^M205 X(\d+)(.*)$**

G-Code to execute:
**M207 X@1**

Regex 2:
**^M204 P(\d+) R(\d+) T(\d+)$**

G-Code to execute:
**M201 X@1 Y@1
M202 X@3 Y@3**

Regex 3:
**^M204 S(\d+)$**

G-Code to execute:
**M201 X@1 Y@1**

closed 2024-01-07 by github-actions[bot]

bsafh, 2024-10-04:
Geeetech A30 (not Pro, M, T, ...) User Repetier style gcode as well. Seems to be no chance to get it work with Orca. Cura has a profile for it, Orca does not!