OrcaSlicer/OrcaSlicer #2183: Fan not turning on, Creality K1

dariotarantini, 2023-09-20:
### OrcaSlicer Version

1.7.0

### OS version

archlinux:latest

### Printer

Creality K1

### How to reproduce

When priting something not all fan turns on after first layer. I usually turn on by hand. 
There could be a fix, if only there was a way to replace Gcode in orcaslicer.  Like following:
[image]

### Actual results

No fan turn on

### Expected results

Fan should turn on

### Project file & Debug log uploads

no log

### Checklist of files to include

- [ ] Log file
- [ ] Project file

closed 2023-12-27 by github-actions[bot]

asmforce, 2024-02-17:
I have the same issue.
After the last K1 firmware upgrade the back fan never starts automatically. I have to start it manually every time.

21IsOnlyTheHalfTruth, 2024-03-22:
Same Issue here with the K1C. But i am kinda new to orca slicer so maybe there is a fix i dont know

allexmail, 2024-07-23:
I have the same problem on k1c. The exhaust fan control does not work, because the names of the fans are mixed up in code M106. I can run it at the beginning of printing via gcode, but it's not convenient.