OrcaSlicer/OrcaSlicer #2186: Internal bridge is critical, it needs its own settings.... Density, Speed, Flow. Its a primary line type for god sake!

PhilBaz, 2023-09-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Internal bridge is a critical feature, and a primary line type like bridge, solid infill, perimeter etc.. 

Depending on model and material it can have very different requirements. Bad internal bridges lead to missed edges and create surface artifacts with burnt discoloration.

I've tested this extensively in Cura that has full internal bridge settings.

Internal bridge needs its own settings.... Density, Speed, Flow

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

All

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Internal bridge settings....Minimum is density, Speed, Flow.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

No options currently. Though I'll have to use Cura for models and materials that have critical internal bridging needs.

**Additional context**
Add any other context or screenshots about the feature request here.

I really feel this is a must. Its like not having a setting for other primary line types like bridge, solid infill, perimeter etc. Internal bridging can be make or break, and is best not left up to chance. I hope this will make it in. People underestimate how critical it is.

AnhDuck, 2023-09-24:
Totally agree with this. There should be more settings for this.

I don't need the fan speed to kick up to 100% on an internal infill bridge.

Lesta99, 2023-10-11:
I also have several complications printing an basic rectangular shape, with two (≈10°) slopes on top that are going inwards and meet in the middle of the object. Just like a "v" if you look at it from the front, only that the "v" is cut out of a rectangular block.
It's nearly impossible to get smooth ans regular layer-lines leading up fron the center on both sides just because it seems that the Infill is being bridged over insufficiently, leading to blobs, wavy layer-lines, generally just anything but a homogeneous slope, just as my way cheaper ender 3 pro via cura printed every time, but my 3x more expensive Qidi X-Max 3 now really struggles with because I choose to support Orcaslicer!
But if this won't get any attention soon, I am pretty much forced to switch back to Qidi Slicer.
In case there would be any wish to receive the object file just get back to me!

Amoulier, 2023-10-30:
The ability the be able to set the internal bridge direction is needed.  is incredible that this simple option is not there to be set!  look this examples!
[image]
[image]

igiannakas, 2023-12-07:
Internal bridge flow parameter has been added in 1.8.1+ In addition the option to disable thick bridge for internal bridges has been added. These two options have the largest impact on external top surface quality.

Wyllio, 2023-12-09:
I'll add that on my large prints, where the infill is finished and moves to internal bridging, which uses the default external bridging fan speed settings, it causes the corners and edges to warp on my ABS/ASA prints. The blasting fan spends more time in the corners while trying to over-cool the internal bridges.

PhilBaz, 2023-12-09:
@Wyllio 

Yeah, Im an ABS guy too. I imagine people who print PLA must think we are crazy for wanting this level of control, but with ABS it is the difference between success and failure. 100%

Eldenroot, 2023-12-13:
> It will still be good to have:
> 
> 1. Internal bridge density
> 2. Internal bridge fan speed
> 3. External bridge fan speed.
> 
> I agree that it would be nice to have some control over the internal bridge direction. Slightly counter intuitive is that it should cross the longest edge. This is because (in line with my testing) internal bridges tend to turn into a surface artifact when they fail/miss the connection to an external wall. Then the hanging end curls up which can 'snow-ball' into a full on charred surface artifact. I've done most of my testing with ABS. the logic may be totally different for TPU.

Yes, needed... anyway we are closer, v1.8.1 added many improvements for this. I hope we can get this soon :)

sleewok, 2024-01-22:
What about the way Cura handles multiple layers following a bridge?  This has a huge impact on bridge quality when printing at higher speeds and acceleration. Going from a bridge to 300mm/s on the next layer is brutal on the bridge.

Here are the settings:
[image]

More details here: https://community.ultimaker.com/topic/22195-introducing-the-experimental-bridging-settings/

Is this related, or should I start another thread?

FaintandFuzzies, 2024-02-14:
I would like to add that the number of wall loops, top shell layers or bottom shell layers shouldn't effect where a bridge is truly needed.  The ultimate for me would be able to paint where bridges are needed (if exposed) similar how manual supports can be painted.

ArahnDoberman, 2024-04-02:
I would love the option to have at least two layers of internal bridging.
With one perpendicular to the other, to have a better anchor for the following layers.

This would come in handy with low infill rates. 
Currently, some islands printed on the internal bridge, get knocked over, as they are only anchored in a single direction
[image]

PhilBaz, 2024-07-03:
Just want to bump this as I'm currently working on a project that requires additional internal bridging control and so I have had to revert to using cura.

Density and angle for internal bridging are critical settings especially for ABS in my testing.

kmarty, 2024-07-16:
Separate settings for "internal bridge cooling".
Especially when printing ASA/ABS you really don't want cooling of "internal bridges". At least not at the same power as "external bridges". Unless you want to have bowed print results.

kmarty, 2024-07-16:
> 
>     1. Internal bridge density
>     2. Internal bridge fan speed
>     3. External bridge fan speed.
> 

Separate settings for "internal bridge fan speed" is must.
Especially when printing ASA/ABS you really don't want cooling of "internal bridges". At least not at the same power as "external bridges". Unless you want to have bowed print results.

jKawka-ITV, 2024-07-25:
Bumping this up as well, this does matter when dealing with ABS/ASA and it seems a bit silly that external and internal bridges have the same fan speed since I'm obviously going to need a fair bit of fan to try bridge 3cm over open air while I don't think internal bridge cooling ever an issue for any infill >= 15%.

mattsiri, 2024-07-30:
Bump. I'm trying to get good top layers in PC atm. The current issue to poor internal bridging. The external bridges look fantastic when using thick bridges at 50% density. It would be really helpful to be able to set the internal bridge density to 50% as well.

Thingatize, 2024-08-10:
> It will still be good to have:
> 
> 1. Internal bridge density
> 2. Internal bridge fan speed
> 3. External bridge fan speed.

4. Internal bridge angle.

pootsy181, 2024-09-04:
bump.... would really like to be able to have internal bridge fan speed.

weeurey, 2024-09-25:
[image]
Bumping, I would really like to set the speed for internal bridging like this here! Or even a "first layer after infill speed" modifier

evgarthub, 2024-10-07:
Today, after 5+ bad results I noticed what causes my model to warp, even with min and max fan thresholds set to 0%. It's because internal infill uses settings for all bridges and overhangs. I noticed how model was printed absolutely fine (ABS) and as soon as internal infill started - 4 seconds and the whole model became detached from the build plate. 
And I was surprised that I cannot change that, I still need overhangs to be cooled, but not internal bridges\infill. 
So here I am, upvoting this feature.

evgarthub, 2024-10-07:
Infill fan speed workaround for now?
- use Cura slicer
- manually remove Set fan speed (M106) from GCODE, you can use ```;TYPE:Internal Bridge``` marker to find all the places. Comment M106 S{speed value} that goes shortly after type comment.

[image]

laukejas, 2024-12-26:
Bump... Having the ability to control the infill direction angle is a must... Right now, I have a part where for some reason the internal infill is at 45°, and there is absolutely no way to make it go straight - even by rotating the part.

[image]

MxBrnr, 2024-12-28:
Here is another from today:
https://discord.com/channels/1137181739773603922/1137184602289610782/1322603105765822477

MxBrnr, 2024-12-28:
Another new request for this from earlier today:
https://discord.com/channels/1137181739773603922/1137241442281074778/1322672285278142474

Eldenroot, 2024-12-28:
Stop spamming! Feel free to push a PR!

MxBrnr, 2025-01-03:
It's not spam. The devs need to know how critical of an issue this is for so any users, and how often. I am simply providing the links to the complaints so that the devs can reference them. I am not able to do the software programming, but if you have that capabiity and this bothers you that much, you could always volunteer to fix it so you don't have to hear it anymore.

igiannakas, 2025-01-03:
I've pushed a PR that enables arbitrary setting of a fan speed for internal bridges.
https://github.com/SoftFever/OrcaSlicer/pull/7906

[image]

Please do understand that the project is supported by a team of volunteers, which do this at their spare time. Sending repeat requests for a feature doesn't make it happen any faster - there are plenty of things that need improvement and not enough time in the evenings/weekends to do them...

Also a kind reminder to support the project by means of a donation through the official support channels below, if you do find the software useful or a specific feature introduced appealing. It helps keep the project alive!  https://github.com/SoftFever/OrcaSlicer?tab=readme-ov-file#support-me

igiannakas, 2025-01-03:
Have also enabled setting the internal bridge angle, overriding the automatically calculated one. PR raised here: https://github.com/SoftFever/OrcaSlicer/pull/7908

Appreciate some testing...

I think that these two PR's should close this issue, correct? We now have independent control of bridge angle, flow, thickness, fan speed for internal bridges implemented or in PR's.

MxBrnr, 2025-01-08:
Looks like the only thing still missing form the original request is an `internal bridge overlap` setting. I would consider this less critical than the others, but maybe that is only because I have not personally experienced any issue with it, yet. We should definitely make sure that internal bridges are properly anchoring to either the adjacent walls, or the wall line of the underlying layer, or both. Having a user-adjustable setting for this overlap would certainly be good so any user with a related issue does have at least one course of action to attempt to remedy.

PhilBaz, 2025-01-08:
@igiannakas Yaaaa! Dreams do come true... Amazing. Thank you!

igiannakas, 2025-01-08:
> Looks like the only thing still missing from the original request is an `internal bridge overlap` setting. I would consider this less critical than the others, but maybe that is only because I have not personally experienced any issue with it, yet. We should definitely make sure that internal bridges are properly anchoring to either the adjacent walls, or the wall line of the underlying layer, or both. Having a user-adjustable setting for this overlap would certainly be good so any user with a related issue (such as bridge edges sagging) does have at least one course of action to attempt to remedy it.

I’ve considered this but orca, prusa and Bambu slicer all extrude an additional perimeter as infill below the internal bridge to assist with anchoring of the internal bridge. Hence there is no concept of overlap as it always overlaps one internal perimeter fully.

igiannakas, 2025-01-08:
> @igiannakas Yaaaa! Dreams do come true... Amazing. Thank you!

Your welcome! The fan speed one was a needed setting. The angle should only be used by exception and only where needed on the model / if needed as orca does a good job of identifying an optimal angle depending on the extrusion direction of the underlying sparse infil.

MxBrnr, 2025-01-10:
> I’ve considered this but orca, prusa and Bambu slicer all extrude an additional perimeter as infill below the internal bridge to assist with anchoring of the internal bridge. Hence there is no [...]

That's even better! Perfect! Thanks!

closed 2025-02-09 by SoftFever