OrcaSlicer/OrcaSlicer #2191: add time estimation for pauses

discip, 2023-09-21:
### Is your feature request related to a problem? Please describe.
To be able to put things like _bearings, magnets, nuts_ ... in to the print, it is essential to know at what place in time that specific layer is reached to put the respective part in and resume the print.
Due to the lack of any time indication, it is currently necessary to estimate when printing will be paused.

### Which printers will be beneficial to this feature
All of them! 😊

### Describe the solution you'd like
1. To address the initial issue we would need separate entries in the **_'Total Estimate - Info Board'_** for each individual pause.
2. Additionally, as proposed here #2045 there could be time estimation per layer as implemented in [PS](https://github.com/prusa3d/PrusaSlicer).
3. And to round things off, this request #679 could be taken into account to make the user's life even easier by letting him know when exactly the pause is going to take place instead of having to calculate that time.

### Describe alternatives you've considered
Remove the upper part above the layer, where the respective part (e.g. nut) should be inserted by cutting it, **`Top shell layers/thickness`** to **`0`**. Calculate the time.

Hope this makes sense! 😃

thanks in advance
kind regards

discip, 2023-09-27:
@niklasb22
I noticed your PR wich addresses to some extent what I'm asking for here.
At least regarding the location of the information.

Do you think this is doable?
If so, may I dare to politely ask whether you could try to implement what is at issue here, as I am not nearly as skilled to be able achieve that?

thanks in advance
kind regards

discip, 2023-09-29:
@Noisyfox
First of all:
I hope you don't mind me tagging you. If you do however, please let me know and I won't do this again.

Do you think this request is feasible?
If so, may I dare to politely ask whether you could try to implement what is at issue here, as I am not able to do this myself?

**By the way: I appreciate your dedication, which is very notable in your attention to detail.** 👍

thanks in advance
kind regards

Noisyfox, 2023-09-29:
I've actually never used the pausing feature so far, but I can see the value of the feature you proposed (and those related issues). Will take a look.

PS: feel free to tag me as you want.

discip, 2023-09-29:
## Thank you very much! Very kind of you! 😃

You should definitely try this! Especially in combination with neodymium magnets!

I am very excited already!

PR #2728 (2023-11-18): Adds overview for custom gcodes like pause, change filament, custom #2191
This inserts an overview of all inserted user-specified gcodes, such as the filament change, pause and custom gcodes.
The layer on which the gcode is located is displayed, as well as a time estimate.
If there is no user-specified gcode in the print, nothing is displayed.

[image]

Closes: #2191
closed 2023-11-18 by SoftFever