OrcaSlicer/OrcaSlicer #2193: Automatic Slicing in Preview Tab

richard-scott, 2023-09-21:
**Is your feature request related to a problem? Please describe.**
When testing out setting changes in the preview tab, I have to click Slice Plate every time

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
I would like to be able to enable auto-slicing, just like UltiMaker Cura has.

**Describe alternatives you've considered**
Clicking the button, but that gets repetitive.

**Additional context**
I've raised this issue here too https://github.com/bambulab/BambuStudio/issues/2542

FinalFrag, 2023-11-04:
You can reslice with CTRL+R, but I agree that auto slicing a la cura would be nice.

mdeeter, 2024-01-23:
Also at https://github.com/SoftFever/OrcaSlicer/discussions/970

closed 2024-05-01 by github-actions[bot]