OrcaSlicer/OrcaSlicer #2194: Implementing Rome (Multi material extruder with single extruder)

Prutsium, 2023-09-21:
Hi,

As we have Rome ( https://github.com/HelgeKeck/rome ) what works with PrusaSlicer & Superslicer but until now not with Orca even not after the recent multi material changes i am looking which variables are exposed as it seems some are not there (yet).

I run into an error because of the variables not known by Orca and get this error:
Failed to generate G-code for invalid custom G-code.
machine_start_gcode Parsing error at line 9: Not a variable name
ROME_START_PRINT EXTRUDER_TEMP=[first_layer_temperature] BED_TEMP=[first_layer_bed_temperature] TOOL=[initial_tool] WIPE_TOWER={wipe_tower} WIPE_TOWER_X={wipe_tower_x} WIPE_TOWER_Y={wipe_tower_y} WIPE_TOWER_WIDTH={wipe_tower_width} WIPE_TOWER_ROTATION_ANGLE={wipe_tower_rotation_angle} COOLING_TUBE_RETRACTION={cooling_tube_retraction} COOLING_TUBE_LENGTH={cooling_tube_length} PARKING_POS_RETRACTION={parking_pos_retraction} EXTRA_LOADING_MOVE={extra_loading_move}

Can you check if it is possible to expose these variables too (they are part of PrusaSlicer)?

Prutsium, 2023-09-21:
Attached PrusaSlicer profile (3mf) but zipped as Git doesnt like 3mf.

[Rome_Shape-Box.zip](https://github.com/SoftFever/OrcaSlicer/files/12686859/Rome_Shape-Box.zip)

Prutsium, 2023-10-11:
Seems to be low prio issue :)
Back to Superslicer then as this project also comes alive (again)

Ocraftyone, 2023-10-11:
Which variable does it say is not recognized?

SirUli, 2024-01-10:
Unfortunately it is still very relevant - would be awesome if this could be implemented.

Prutsium, 2024-01-10:
> Unfortunately it is still very relevant - would be awesome if this could be implemented.

Nope ... i got it working See the Discord channel: https://discord.com/channels/1137181739773603922/1150397562843582534/1190326061037277214

closed 2024-04-19 by github-actions[bot]