OrcaSlicer/OrcaSlicer #2195: Support Critical Regions Only No longer performs as expected

anthonyp-cns, 2023-09-21:
### OrcaSlicer Version

1.7.0 

### OS version

Windows 10 22H2

### Printer

Bambu Lab X1C

### How to reproduce

Enable support
Enable support critical regions only

### Actual results

The critical regions only option appears to have no effect. 

Version 1.6.6 performed as expected supporting only critical regions, while 1.7.0 adds support everywhere

### Expected results

Support of only critical regions instead of full model

### Project file & Debug log uploads

[Honeycomb Cup - orca 1.7.0 critical regions support issue.zip](https://github.com/SoftFever/OrcaSlicer/files/12689233/Honeycomb.Cup.-.orca.1.7.0.critical.regions.support.issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2023-09-22:
The default style tree is broken in 1.7, switch to a non-default style (i.e., TreeHybrid) instead.

anthonyp-cns, 2023-09-24:
Wow you are on top of it! Thanks for the quick response and submitting the code to fix this. 

I am going to close this bug. 

Thanks again!

closed 2023-09-24 by anthonyp-cns