OrcaSlicer/OrcaSlicer #2197: Calibration menu shows ">No printer", can't seem to use Calibration menu

cjshaker, 2023-09-21:
### OrcaSlicer Version

1.7.0

### OS version

Windows 11 Home 22H2 build 22621.2283

### Additional system information

Device name	cjshaker
Processor	11th Gen Intel(R) Core(TM) i7-11700K @ 3.60GHz   3.60 GHz
Installed RAM	16.0 GB (15.8 GB usable)
Device ID	B5CDE8A9-447B-48FE-B136-D7E01C09CC4B
Product ID	00325-82189-88851-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

Flsun V400, and Prusa MK3S

### How to reproduce

I'm a new Orca Slicer user. Apologize in advance if this is just operator error, but the wiki doesn't appear to show this?

Click 'New Project',  select Flsun V400 printer, select a filament (like FlSun Generic PLA), select 0.20mm Standard@FLSun V400 for print quality, select 'Preview' tab - all looks good. Now, select 'Calibration' tab to the right of 'Project', and nothing works:
[image]
[image]
[image]

ALSO, the 'Project' tab says 'no model information'. Is this expected?
[image]

I tried the same thing with the Prusa i3 MK3 printer, and that 'Calibration' tab is also greyed out for that printer too:
[image]
[image]

Either I'm missing something in the user interface, or this 'Calibration' tab is broken in 1.7.0?

Hmm. I just noticed the "Calibration" in the TOP menu bar, and that appears to work. Why are there TWO ways of doing Calibration?

### Actual results

Can't select 'Manual Calibration' nor 'Auto Calibration'. Calibration says "> No printer". Screenshots above.

### Expected results

I thought I used to be able to launch calibration in a previous version of Orca slicer?

### Project file & Debug log uploads

[Project.zip](https://github.com/SoftFever/OrcaSlicer/files/12693874/Project.zip)
[debug_Thu_Sep_21_14_45_00_3296.log.zip](https://github.com/SoftFever/OrcaSlicer/files/12693875/debug_Thu_Sep_21_14_45_00_3296.log.zip)
[OrcaSlicer.conf.zip](https://github.com/SoftFever/OrcaSlicer/files/12693877/OrcaSlicer.conf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2023-09-22:
Because the calibration tab is added in BambuStudio and only works for their printers, while the one on menu bar is the Orca's.

YELLOWFROG0623, 2023-09-22:
Hi!
Sorry for cutting in.
I went through the same process of trial and error for about 2 days, but I finally solved the mystery and felt refreshed.
thank you very much.

DrDrache, 2023-09-28:
can we get an option to remove (hide) the tab for non-bamboo printers? or perhaps get it working for others?

stopanko, 2023-09-29:
So confusing. 
[image]
Bambu Calibrationts. Works only with Bambu printers.

[image]
Orca calibrations. Free to use))

Noisyfox, 2023-12-29:
Should have been fixed a while ago by hiding the bambu calibration tab for non-bambu printers.

closed 2023-12-29 by Noisyfox