OrcaSlicer/OrcaSlicer #2200: Option to place gcode preview text block at the end of file, rather than beginning.

PhilBaz, 2023-09-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

An option to place gcode preview text block at the end of file, rather than beginning. For those of us who look at the gcode it would be much nicer to not have to navigate past the huge preview block at the beginning.

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

all gcode peepers.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

An option to place gcode preview text block at the end of file, rather than beginning

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

za-x-az, 2023-09-29:
Yes this please! 🙏

PhilBaz, 2023-12-29:
Bump for git.....would be nice!

za-x-az, 2024-03-29:
Any chance we will get this?

PhilBaz, 2024-03-29:
@za-x-az  Would be nice wouldn't it?  it will be interesting to try the new gcode editor in 2.0. Maybe that will either make the need for this feature more relevant? Don't know, we will have to see.

za-x-az, 2024-06-28:
> @za-x-az Would be nice wouldn't it? it will be interesting to try the new gcode editor in 2.0. Maybe that will either make the need for this feature more relevant? Don't know, we will have to see.

bump? :D

closed 2024-10-05 by github-actions[bot]

PhilBaz, 2024-10-08:
Bump....would be nice.